ArminJo/ServoEasing #1: ESP8266 and STM32

TByte007, 2019-05-01:
Do you think it would be reasonably easy to port this library to ESP8266 and STM32 ? This is way way way too good library to have it only for ATMEL :)

Regards

eried, 2019-05-02:
I agree with @TByte007 :/ I tried to remove the incompatible parts but I think it is not that simple (I just wanted a simple easing of movements)

ArminJo, 2019-05-02:
So I will try to port the lib, but it may take several weeks.
Maybe I can ask you to test the port before releasing it?
How can I contact you, I have no facebook etc :-).

eried, 2019-05-02:
If you create a branch, I can test it and report here :)

ArminJo, 2019-05-09:
> If you create a branch, I can test it and report here :)

The ESP8266 version is now ready for testing at the master branch.

eried, 2019-05-10:
Cool! thanks! it compiles but it does not appear in the compatible libraries without https://github.com/ArminJo/ServoEasing/pull/3

I am going to test it with a servo after holidays :)

TByte007, 2019-07-07:
Wow thank you VERY much ! :) 
Do you think that it will work on STM32F1 (bluepill) too as it is based on ARM (as ESP8266) too ?

ArminJo, 2019-07-08:
I will check it, but be aware, ESP8266 is an Xtensa core  and not ARM Architecture.

TByte007, 2019-07-09:
> I will check it, but be aware, ESP8266 is an Xtensa core and not ARM Architecture.

I messed up something in my reading somewhere along the lines I was probably thinking of 32bit RISC :)

ArminJo, 2019-07-11:
Hi TByte007,
The latest version now compiles for me for a STM32F103.
Can you test it on real Hardware?
Thanks

TByte007, 2019-07-15:
> Hi TByte007,
> The latest version now compiles for me for a STM32F103.
> Can you test it on real Hardware?
> Thanks

Yes, but in a week when I get back from the vacation. Thank you !

ArminJo, 2019-09-19:
How was you vacation?

TByte007, 2019-09-20:
Well, fun - but they managed to find me a project which "should have been
finished yesterday"
 (tm) in the middle of it (talking traffic lights button for blind people
with IR non-contact button and vibration motor and tow how power LEDs0...)
so I was rather occupied the last month+ :)
I'll test the lib tonight - since yesterday I'm back on my robot project
(which involves both STM32 and ESP8266) oh and a PWM multiplier (PCA9685)
which will be fun to see if it will work with the lib too :)

TByte007, 2019-10-01:
Sorry for the delay - the new core had problems with the USB (serial)
Managed to test it and the blocking part works but the part where it uses interrupts doesn't.

ArminJo, 2019-10-19:
Hi,
I ported it to STM and it  runs on my BluePill board.
Does it also work for you?

ArminJo, 2019-11-12:
Released new version 1.4.1 with BluePill support.

closed 2019-11-12 by ArminJo