ArminJo/ServoEasing #5: ESP32

botbuilder1, 2019-06-08:
I get an error compiling for a ESP32:

"This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."

Are there any plans on including the ESP32?

ArminJo, 2019-06-08:
No one has asked for this before. For what purpose do you need it on the ESP32?

veitk, 2019-06-08:
i didn't tried it yet but i also wanted to use this on the esp32 to replace the esp32servo lib.

this would be really awesome if you can support esp32 :)

botbuilder1, 2019-06-08:
> No one has asked for this before. For what purpose do you need it on the ESP32?

The servo motion for my ice fishing jigger is so jerky.  Would LOVE to have that bounce built into it too.

TIA

ArminJo, 2019-06-08:
OK let me have a look, if I can use the esp32servo lib on a ESP32.

ArminJo, 2019-06-08:
OK guys, please download and test it.

botbuilder1, 2019-06-11:
> OK guys, please download and test it.

Seems to work well, thanks!

ArminJo, 2019-06-11:
Thanks, I now created a new release, so that the port may be soon available at the Arduino IDE.

closed 2019-06-11 by ArminJo