ArminJo/ServoEasing #6: Cannot detach() a servo?

boomninjavanish, 2019-06-22:
Hey! First off, thanks for making such a nice library! I am using it well on ESP8266 hardware. However, I wish to detach servos in my project and could not find the detach() method in ServoEasing. 

I resorted to using a hacky solution that involves declaring more servos using the Servo library, attaching them using the same pins that were used with ServoEasing library , then detaching those servos. I then reattach the servos using the ServoEasing objects when I am ready to use the servos again.

This isn't the best way to conserve memory and looks terrible in my code. Do you plan on implementing the detach() method into ServoEasing? If not, do you have any other suggestions? 

I am trying to use cheap servo motors (SG90s) and the unit will be powered on 24/7 for 3 weeks. The servos will be activated once a day for 5 minutes. I am using detach() to avoid jitter of the servos and prevent problems from possibly being bumped when not in use.

Thanks!

ArminJo, 2019-06-24:
Hello,
thank you very much for your nice feedback!
And thank you for describing your problem, so I was highly motivated to add the detatch method.
Is this what you need or do you need a true destructor?
And can you please test it?

Good luck
Armin

boomninjavanish, 2019-06-24:
Hi Armin!

Thanks for adding it! I tested and it works great! For some reason, your detach() is much smoother than the Servo library's. It was kinda jerky when detaching before.

For this project, I do not see the need for a true destructor as the whole microcontroller isn't needing the extra memory between the detach() and attach() methods. In fact, the project just polls for input until the attach() occurs again.

Thanks again for your help! This made my project SO much easier to implement!

ninja

closed 2019-06-24 by boomninjavanish