ArminJo/ServoEasing #7: Immediately stop servo?

veitk, 2019-07-08:
Hi Armin, 

i love how my little robot moves now thanks to your library. 

But: i need to stop movements sometimes immediately and move to another direction. I tried to use "stopAllServos()" but the movement is still completed and then the next command is processed. 

Do you have an idea to solve this? thanks!

ArminJo, 2019-07-08:
Hi Veitk, thank you for your feedback.
Stop all servos works reliable for all my examples. 
Maybe your code just issues the next command after stopping, so that stopping has no effect.
Did you use the latest version from Github?

veitk, 2019-07-14:
Thanks, i didn't have time before ... and yes, i use the latest version from github.

I did a quite simple thing in a loop to test it. 
If the servo is moving i set result to true (with infrared remote), it should stop and toggle direction.
But it finishes the movement, then moves the other way....
```
if (result == true) {
stopAllServos();
                if (toggle== true){ 
                  toggle = false;   
                  servoBody.startEaseToD(0, 1000);  
                  while (servoBody.isMoving()) {
                      ;
                  }
                } else {
                  toggle = true;
                  servoBody.startEaseToD(180, 1000);
                  while (servoBody.isMoving()) {
                      ;
                  }
                }
```

ArminJo, 2019-07-15:
Hi, this is an endless loop and it will run until the movement has finished!
>                   while (servoBody.isMoving()) {
>                       ;
>                   }
try while (servoBody.isMoving() && <direction is not toggled>

closed 2019-07-15 by ArminJo