ArminJo/ServoEasing #9: ESP32: 1 of 2 servo always moves to 90° and back on startup

veitk, 2019-09-22:
Hi, 
my little robot uses two servos on an ESP32. 
On power on one of the servos rotates 90° and then back to almost 0. 
After that the code reaches the position where the servo is attached and get the command to move to 0 which it then does. 
The second servo is perfectly still on startup. 
This is happening on all my bots all the time. Do you think this is a problem with the pin on the ESP32 or do you have any other explanation?

The moving servo is on pin 19, the other one on pin 18.

ArminJo, 2019-09-22:
Pls attach you code.
What do you mean with 
> the position where the servo is attached
and
> The second servo is perfectly still on startup.
Is this expected or not?

veitk, 2019-09-27:
As i'm writing this answer i maybe realise what the problem is. 

Arm servo should be at 0, body servo at 90. 
Thats the position it has when the bot is turned off. In my first post i wrote it jumps to 90 and then to 0.

On attach the body servo jumps to 0. Then it eases to 90 which it says in my code. 
It then makes another small jump by 5 degrees. 
This happens all the time.
Would be nice if the servo simply does nothing on attach - but i guess this is not possible...

ServoEasing servoArm;
ServoEasing servoBody;

servoArm.attach(PIN_SERVO_ARM);
servoArm.setEasingType(EASE_CUBIC_IN_OUT);
servoArm.startEaseTo(0, 30); 
servoBody.attach(PIN_SERVO_BODY);        
servoBody.setEasingType(EASE_CUBIC_IN_OUT);       
servoBody.startEaseTo(90, 30);

ArminJo, 2019-09-28:
Thank you for focusing my attention to this point.
I updated all examples and the readme.
The point is, that ServoEasing cannot guess which angle your physical servo has at startup time. 
So it always starts at 0 degree unless you state **like in ALL examples** 
`servoArm.write(<your start angle>);`
`servoBody.write(<your start angle>);`

But the jump by 5 degree should not happen. I have observed this before on a cheap SG90 servo, whose internal control was quite inaccurate.

veitk, 2019-09-28:
perfect, thank you.

closed 2019-09-28 by veitk