ArminJo/ServoEasing #10: Feature Requst: additional feature for .attach

sapics, 2019-10-18:
Thanks for very useful library!

I would like to use `attach` as follows.
`.attach(pin, minPulse, maxPulse, degreeForMinPulse, degreeForMaxPulse)`
Because some of servos move range are not 180 degree.

If the below examples are possible, it looks very useful.
```c++
.attach(5, 500, 2500); // default values (0, 180) looks good.
.attach(5, 500, 2500, -60, 60);
.attach(5, 500, 2500, 90, -30);
```

ArminJo, 2019-10-19:
Hi sapics,
the new feature is now available, please test it and give feedback.

sapics, 2019-10-20:
Thank you very much ArminJo!

It looks great!
I ordered servo motors, and it will arrive in a few days.
I will make a test report as soon as possible!

sapics, 2019-10-21:
It works well as I expected in Arduino UNO.
Thanks again!

closed 2019-10-21 by sapics