ArminJo/ServoEasing #11: Continuous Rotation Servos

Eebel, 2019-10-18:
Hi, I have full sized R5-D4 Astromech (sorta like R2-D2) which has three Adafruit PWM expanders currently.  I am updating the sketch to use your ServoEasing library.  I already added the NeoPatterns library to run the NeoPixels used in his lighting and I want to further refine the sketch with multiple servo sequences across his dome and body.  One issue I think I have is the Servo Easing library does not seem to support continuous rotation servos.  I have one in my build that runs a tool in and out on a linear rail (the Buzz Saw).  Here is what I have working without your library: https://youtu.be/o_OPfRII7wM.  

is it possible to get support for continuous rotation servos?  Or if supports these servos now, how do I do it?

Thanks

Eebel

ArminJo, 2019-10-19:
Hi Eebel,
Impressive work!

I added a new example for usage with continuous rotating servo. Can you test it?
Does it work with your servo?
Thanks
Armin

Eebel, 2019-10-19:
I'll take a look.  Thanks!

Eebel

Eebel, 2019-10-19:
I get this error when trying to compile.

/Users/MiniMacBookPro/Documents/Arduino/libraries/ServoEasing/src/ServoEasing.cpp: In function 'void enableServoEasingInterrupt()':
/Users/MiniMacBookPro/Documents/Arduino/libraries/ServoEasing/src/ServoEasing.cpp:837:2: error: 'TCCR1' was not declared in this scope
  TCCR1
  ^~~~~
/Users/MiniMacBookPro/Documents/Arduino/libraries/ServoEasing/src/ServoEasing.cpp:837:2: note: suggested alternative: 'TCCR1A'
  TCCR1
  ^~~~~
  TCCR1A

ArminJo, 2019-10-20:
Which board did you specify in arduino?

Eebel, 2019-10-20:
I am using the Mega2560.

Eebel

ArminJo, 2019-10-27:
Since for me it compiles for a Mega2560, I close the Issue.

closed 2019-10-27 by ArminJo

Eebel, 2019-10-28:
There is an uncommented TODO variable (it seems) causing the issue.  It only shows up if you are using the expander board.  When I comment out TCCR1, the code compiles.

void enableServoEasingInterrupt() {
#if defined(__AVR__)
#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
#if defined(USE_PCA9685_SERVO_EXPANDER)
	// TODO convert to timer 5
	TCCR1
	// set timer 1 to 20 ms
	TCCR1A = _BV(WGM11);// FastPWM Mode mode TOP (20 ms) determined by ICR1 - non-inverting Compare Output mode OC1A+OC1B
	TCCR1B = _BV(WGM13) | _BV(WGM12) | _BV(CS11);// set prescaler to 8, FastPWM mode mode bits WGM13 + WGM12
	ICR1 = 40000;// set period to 20 ms
#endif

ArminJo, 2019-10-28:
Oh sorry and thank you.
It should be fixed now. Please test.

Eebel, 2019-10-28:
OK, it compiles now.  I'll test it with the example code.  IT will take me a while to get to test it on my robot as it is a massive refactoring to get things running again.

Thanks!

Tim

ArminJo, 2019-10-28:
Good luck!!!
Armin

commit e3783c52 (2019-10-28): Timer 5 for ATmega2560 and USE_PCA9685_SERVO_EXPANDER
Closes #11
commit cdc14de5 (2019-10-28): Timer 5 for ATmega2560 and USE_PCA9685_SERVO_EXPANDER
Closes #11
commit 16bdf1d7 (2019-10-28): Timer 5 for ATmega2560 and USE_PCA9685_SERVO_EXPANDER
Closes #11