ArminJo/ServoEasing #12: Multiple PCA9685 Expanders

Eebel, 2019-11-01:
I can seem to get one PCA9685 expander to work at a time.  I have three in my robot.  Maybe I am setting them up incorrectly.

I initialize the servo objects  in the beginning of the sketch:

      ServoEasing pwm1_Servo4(0x40, &Wire); //Data Panel Door
      
      ServoEasing pwm2_Servo1(0x41, &Wire); //BuzzSlide
      ServoEasing pwm2_Servo2(0x41, &Wire); //Motivator
      ServoEasing pwm2_Servo3(0x41, &Wire); //Zapper 3-7
      ServoEasing pwm2_Servo4(0x41, &Wire); //Left BreadPan Door
      ServoEasing pwm2_Servo6(0x41, &Wire); //RightBread Pan Door
      ServoEasing pwm2_Servo7(0x41, &Wire); //CBI Door
      ServoEasing pwm2_Servo8(0x41, &Wire); //Lower Utility Arm
      ServoEasing pwm2_Servo9(0x41, &Wire); //Upper Utility Arm
      ServoEasing pwm2_Servo11(0x41, &Wire); //Buzz Door
      ServoEasing pwm2_Servo12(0x41, &Wire); //GripperLift
      ServoEasing pwm2_Servo13(0x41, &Wire); //Gripper Tool
      ServoEasing pwm2_Servo14(0x41, &Wire); //Interface Lift
      ServoEasing pwm2_Servo15(0x41, &Wire); //Interface Tool
      
      ServoEasing pwm3_Servo1(0x42, &Wire); //Pie 1
      ServoEasing pwm3_Servo2(0x42, &Wire); //Pie 2
      ServoEasing pwm3_Servo3(0x42, &Wire); //Pie 3
      ServoEasing pwm3_Servo4(0x42, &Wire); //Pie 4
      ServoEasing pwm3_Servo5(0x42, &Wire); //Pie 5
      ServoEasing pwm3_Servo6(0x42, &Wire); //Pie 6
      ServoEasing pwm3_Servo7(0x42, &Wire); //Pie 7
      ServoEasing pwm3_Servo8(0x42, &Wire); //LifeSlide
      ServoEasing pwm3_Servo9(0x42, &Wire); //LifeTurn

I call this in the Setup() of the sketch:

     pwm1_Servo4.attach(pwm1_4Info.pinNum, 2400, 620);//DataPanel Door
     pwm1_Servo4.write(pwm1_4Info.minDeg); //
    
      pwm2_Servo1.attach(pwm2_1Info.pinNum, 620, 2400);//Buzz Slide
      pwm2_Servo1.write(pwm2_1Info.minDeg); //     
      pwm2_Servo4.attach(pwm2_4Info.pinNum, 620, 2400);//Gripper Door
      pwm2_Servo4.write(pwm2_4Info.minDeg); //     
      pwm2_Servo6.attach(pwm2_6Info.pinNum, 2400, 620);//Interface Door
      pwm2_Servo6.write(pwm2_6Info.minDeg); //     
      pwm2_Servo7.attach(pwm2_7Info.pinNum, 2400, 620);//CBI Door
      pwm2_Servo7.write(pwm2_7Info.minDeg);  
      pwm2_Servo8.attach(pwm2_8Info.pinNum, 620, 2400);
      pwm2_Servo8.write(pwm2_8Info.minDeg);
      pwm2_Servo9.attach(pwm2_9Info.pinNum, 2400, 620);;
      pwm2_Servo9.write(pwm2_9Info.minDeg);       
      pwm2_Servo11.attach(pwm2_11Info.pinNum, 2400, 620);//Buzz Door
      pwm2_Servo11.write(pwm2_11Info.minDeg); //      
      pwm2_Servo12.attach(pwm2_12Info.pinNum, 620, 2400);//Gripper Lift
      pwm2_Servo12.write(pwm2_12Info.minDeg); //     
      pwm2_Servo13.attach(pwm2_13Info.pinNum, 620, 2400);//Gripper
      pwm2_Servo13.write(pwm2_13Info.minDeg); //    
      pwm2_Servo14.attach(pwm2_14Info.pinNum, 2400, 620);
      pwm2_Servo14.write(pwm2_14Info.minDeg); //     
      pwm2_Servo15.attach(pwm2_15Info.pinNum, 620, 2400);
      pwm2_Servo15.write(pwm2_15Info.minDeg); //

      pwm3_Servo1.attach(pwm3_1Info.pinNum, 2400, 620);
      pwm3_Servo1.write(pwm3_1Info.minDeg); 
      pwm3_Servo2.attach(pwm3_2Info.pinNum, 2400,620);
      pwm3_Servo2.write(pwm3_2Info.minDeg); 
      pwm3_Servo3.attach(pwm3_3Info.pinNum, 2400,620);
      pwm3_Servo3.write(pwm3_3Info.minDeg); 
      pwm3_Servo4.attach(pwm3_4Info.pinNum,2400,620);
      pwm3_Servo4.write(pwm3_4Info.minDeg); 
      pwm3_Servo5.attach(pwm3_5Info.pinNum, 2400,620);
      pwm3_Servo5.write(pwm3_5Info.minDeg); 
      pwm3_Servo6.attach(pwm3_6Info.pinNum, 2400,620);
      pwm3_Servo6.write(pwm3_6Info.minDeg); 
      pwm3_Servo7.attach(pwm3_7Info.pinNum, 2400, 620);
      pwm3_Servo7.write(pwm3_7Info.minDeg); 

When I run this code,  only the first pwm driver works.  By commenting out whole sections of the attach commands, I can get each individual driver to run its servos.  However, I cannot get all three working at one time.  Can you assist?

Tim

ArminJo, 2019-11-02:
Hi Tim, 
can you try to insert after your first attach: // pwm1_Servo4.attach(pwm1_4Info.pinNum, 2400, 620);//DataPanel Door

.... insert begin
 Wire.begin();
    Wire.setClock(800000);// 1000000 does not work - maybe because of parasitic breadboard capacities
	// Send software reset to expander
    Wire.beginTransmission(PCA9685_GENERAL_CALL_ADDRESS);
    Wire.write(PCA9685_SOFTWARE_RESET);
    Wire.endTransmission();
	// set to 20 ms period
    pwm1_Servo4.I2CWriteByte(PCA9685_MODE1_REGISTER, _BV(PCA9685_SLEEP));// go to sleep
    pwm1_Servo4.I2CWriteByte(PCA9685_PRESCALE_REGISTER, PCA9685_PRESCALER_FOR_20_MS);// set the prescaler
    pwm1_Servo4.I2CWriteByte(PCA9685_MODE1_REGISTER, _BV(PCA9685_AUTOINCREMENT));// reset sleep and enable auto increment

    pwm2_Servo1.I2CWriteByte(PCA9685_MODE1_REGISTER, _BV(PCA9685_SLEEP));// go to sleep
    pwm2_Servo1.I2CWriteByte(PCA9685_PRESCALE_REGISTER, PCA9685_PRESCALER_FOR_20_MS);// set the prescaler
    pwm2_Servo1.I2CWriteByte(PCA9685_MODE1_REGISTER, _BV(PCA9685_AUTOINCREMENT));// reset sleep and enable auto increment

    pwm3_Servo1.I2CWriteByte(PCA9685_MODE1_REGISTER, _BV(PCA9685_SLEEP));// go to sleep
    pwm3_Servo1.I2CWriteByte(PCA9685_PRESCALE_REGISTER, PCA9685_PRESCALER_FOR_20_MS);// set the prescaler
    pwm3_Servo1.I2CWriteByte(PCA9685_MODE1_REGISTER, _BV(PCA9685_AUTOINCREMENT));// reset sleep and enable auto increment
... insert end

If this works, I know how to modify the example and the library to support multiple Expanders.
Thanks
Armin

Eebel, 2019-11-02:
OK, I'm on it.  Will report back.

Eebel, 2019-11-02:
Hmm,

When I insert the code it stops all communication.  No servos work.

Tim

closed 2019-11-02 by ArminJo

ArminJo, 2019-11-02:
Hi Tim, 
I have changed ServoEasing.cpp and ServoEasing.h and now it does a reset only once, but an initalisation at every attach(). 
It would be very nice, if you can test it, especially if the initialization at each attach() disturbes the servos already attached. If yes, I have to add code to do the initialisation only once for each expander, but I hope, I can avoid it.

And just for my curiosity... did you change MAX_EASING_SERVOS at line 58 in ServoEasing.h and what IDE do you use? Did you try [sloeber](http://eclipse.baeyens.it/stable.php?OS=Windows)? 

Thanks a lot for helping to improve the library!
Armin

Eebel, 2019-11-02:
I did a quick test with the new version and it all seems to work well.  I did not change the MAX_EASING_SERVOS as it said it was per expander so the 16 value was appropriate.

I am currently using the Arduino IDE.  I have been using XCODE for viewing your library code.  I will investigate sloeber as I don't really like the Arduino IDE.

Thanks for the tip on the IDE and thanks for being so responsive in your updates.  
I glad I can contribute in a small way.

Tim

ArminJo, 2019-11-02:
To clarify the MAX_EASING_SERVOS issue, I added the following remark to the source.
```
 * If the value of MAX_EASING_SERVOS is smaller than the amount of servos declared,
 * attach() will return error and detach() will not work as expected.
 * As well as all *AllServos*() functions and isOneServoMoving()
 * won't work correctly! (they will only work for the first MAX_EASING_SERVOS -2 servos)
 *
 * If you do not need these functions, you may define MAX_EASING_SERVOS as 1
```
So consider to change the value or get the latest code, which now includes a bugfix for "_wrong_" MAX_EASING_SERVOS declaration.
You are the first one with more than 16 servos, so it takes a while to get the software bug free.
Thanks

commit 0cacfa71 (2019-11-02): Version 1.4.0 Support for multiple PCA9685 expander.
Closes #12