ArminJo/ServoEasing #13: Continuous Rotation Turns wrong way initially

Eebel, 2019-11-08:
Hi,

I have the following test code simulating a door opening, a tool moving and then the continuous rotation servo extending a buzz saw on a linear rail, when I push a button.  When I push a second button, the process reverses.  However, the continuous rotation servo initially turns the wrong way for a bit before reversing to the proper direction.  I am using a limit switch in the design so the servo doesn't bottom out and grind when the buzz saw is fully retracted and so it doesn't move during the initialization process when I first start the droid if the buzz saw is fully retracted.

Am i calling the functions wrong to cause this behavior?

Tim

/*
 * This routine uses button switchs to simulate commands sent by a bluetooth controller.
 * 
 * 
 * 
 * 
 */
//2016.12.08

#include "ServoEasing.h"

int ledPin = 5;
int buttonApin = 9;
int buttonBpin = 8;
int limitSwitchPin = 22;

const int SERVO1_PIN = 0;
const int SERVO2_PIN = 1;
const int SERVO3_PIN = 2;

ServoEasing Servo1(PCA9685_DEFAULT_ADDRESS, &Wire);
ServoEasing Servo2(PCA9685_DEFAULT_ADDRESS, &Wire);
ServoEasing Servo3(PCA9685_DEFAULT_ADDRESS, &Wire);
ServoEasing Servo11(PCA9685_DEFAULT_ADDRESS, &Wire);

byte leds = 0;

    struct ServoInfo {
    uint16_t minDeg;
    uint16_t maxDeg;
    int pinNum;
    int pwmNum;
    String servoName;
    };

ServoInfo S1_Info;
ServoInfo S2_Info;
ServoInfo S3_Info;
ServoInfo S11_Info;

void setup() 
{
  pinMode(ledPin, OUTPUT);
  pinMode(buttonApin, INPUT_PULLUP);  
  pinMode(buttonBpin, INPUT_PULLUP);  
  pinMode(limitSwitchPin, INPUT_PULLUP);

  S1_Info.servoName = "GripperDoor";
  S1_Info.minDeg = 90;
  S1_Info.maxDeg = 0;
  S1_Info.pwmNum = 1;
  S1_Info.pinNum  = 0;
  S2_Info.servoName = "GripperLifter";
  S2_Info.minDeg = 68;
  S2_Info.maxDeg = 130;
  S2_Info.pwmNum = 1;
  S2_Info.pinNum = 1;
  S3_Info.servoName = "Gripper";
  S3_Info.minDeg = 72;
  S3_Info.maxDeg = 120;
  S3_Info.pwmNum = 1;
  S3_Info.pinNum = 2;
  S11_Info.servoName = "BuzzSlide";
  S11_Info.minDeg = 100;
  S11_Info.maxDeg = -100;
  S11_Info.pwmNum = 1;
  S11_Info.pinNum  = 11;

    // Attach servos to pins
    Serial.println(F("Attach servos"));

    Servo1.attach(S1_Info.pinNum, 544,2400);//gripper door
    Servo2.attach(S2_Info.pinNum, 544, 2400);//gripper lifter
    Servo3.attach(S3_Info.pinNum, 2400,544);//gripper
    Servo11.attach(S11_Info.pinNum, 544, 2400, -100, 100);//BuzzSlide

    // Set servos to start position. 
    Servo1.write(S1_Info.minDeg);
    Servo2.write(S2_Info.minDeg);
    Servo3.write(S3_Info.minDeg);
    if (digitalRead(limitSwitchPin) == HIGH){
      Servo11.write(100); 
    }
    
    setSpeedForAllServos(600);
    // Just wait for servos to reach position
    delay(500);

   //Silence the servos
    Servo1.detach();
    Servo2.detach();
    Servo3.detach();
    Servo11.detach();
  
}

void loop() 
{
  if (digitalRead(buttonApin) == LOW)
  {
    digitalWrite(ledPin, HIGH); 
    Servo3.easeToD(S3_Info.maxDeg,200);
    Servo2.easeToD(S2_Info.minDeg,200);
    Servo1.easeToD(S1_Info.minDeg,200);
    Servo11.startEaseTo(100);//<---initially turns wrong way here, then works correctly
    Servo1.detach();
    Servo2.detach();
    Servo3.detach();
  }
  if (digitalRead(buttonBpin) == LOW)
  {
    digitalWrite(ledPin, LOW);
    Servo1.easeToD(S1_Info.maxDeg,300);//door
    Servo2.easeToD(S2_Info.maxDeg,300);//lifter
    Servo3.easeToD(S3_Info.maxDeg,150);
    Servo3.easeToD(S3_Info.minDeg,150);
    Servo3.easeToD(S3_Info.maxDeg,150);
    Servo3.easeToD(S3_Info.minDeg,150);
    Servo3.easeToD(S3_Info.maxDeg,150);
    Servo3.easeToD(S2_Info.minDeg,150);
    Servo11.write(-100);
    delay(1500);
    Servo1.detach();
    Servo2.detach();
    Servo3.detach();
    Servo11.detach();
  }
  if (digitalRead(limitSwitchPin) == HIGH){
     //Run the servo until it reaches the limit
      Servo11.update();
    }
    else{
     //limit is reached, stop the servo
      Servo11.detach();
    }
}

ArminJo, 2019-11-09:
Hi Eebel,
thanks for forcing me to improve my example!

I just uploaded it together with ServoEasing.h.
The problem is likely, that your servos STOP value is not **exactly 1500 microseconds**.
If the stop value of your servo is NOT exactly 1500 microseconds, you must take the new ServoEasing.h file and change the value of MICROSECONDS_FOR_ROTATING_SERVO_STOP.
You should **not** use the 544 and 2400 values for it. See example and definitions in ServoEasing.h.

Thanks 
Armin

Ps. Speed of 600 for all servos seems a bit high, since the maximum physical speed of my servos is 350. Just use 150 and you can omit the 150 in `Servo3.easeToD(S2_Info.minDeg,150);`

And instead of 
```
Servo3.attach(S3_Info.pinNum, 2400,544);//gripper
```
you may use 
```
Servo3.attach(S3_Info.pinNum, 544, 2400);//gripper
Servo3.setReverseOperation(true);
```

Eebel, 2019-11-09:
Thanks for the feedback.  I misunderstood the easeToD command.  I though the D part was a delay/time period for the movement.  I did not understand it was a speed.  I'll adjust my code accordingly.  The only question I have is if I have two continuous rotation servos that have different values for the not moving, wouldn't one of them be wrong if there is only one value for all of them?  Or are certain brands close enough that if you have one of that brand then another will behave the same.  I don't know as I haven't added the second continuous rotation servo to my droid yet.

Tim

ArminJo, 2019-11-09:
OK, I missed that you use the easeTo**D**() command and you were totally right with the assumption, that the parameter was a delay. Sorry.
For the easeTo() function, the parameter is speed (degree per seconds) and this speed is constant regardless of the size of motion, which is the reason I use it always.

The problem with the rotational servo is to know at which value it is stopping. This value must be the value chosen if you call write(0); 
Best way is to use a min value as `stop value - delta`  and a max value as `stop value + delta` and a symmterical mapping like ,`100, -100` - see [example](https://github.com/ArminJo/ServoEasing/blob/master/examples/ContinuousRotatingServo/ContinuousRotatingServo.ino),
With this in mind you can use different stop values for every servo, you do not need to use the `#defines`, the are only for convenience.

Armin

Eebel, 2019-11-09:
Thanks for taking the time to explain this!  It will really allow me to get the servo animation of my droid to the next level!

Tim

Aren't you impressed?

ArminJo, 2019-11-09:
Glad to hear :-)

closed 2019-11-09 by ArminJo