ArminJo/ServoEasing #14: Servos not waiting their turn

Eebel, 2019-11-21:
Hi, I've found an odd behavior.  I am using the following code to open pie panels in sequence.  The first four open as expected and the last three open at the same time instead of sequentially.  The other PWM expanders work as expected in the following code.  However, I have had other bits of code where I have had to insert delay(300); to keep the sequence of the servos properly timed.  I just chose this example because it is easy to see.

https://youtu.be/lQYNI1CdSG0

Here is the code:
      pwm3_Servo1.easeTo(pwm3_1Info.maxDeg);//Pie1
      pwm3_Servo2.easeTo(pwm3_2Info.maxDeg);//Pie2
      pwm3_Servo3.easeTo(pwm3_3Info.maxDeg); //Pie3
      pwm3_Servo4.easeTo(pwm3_4Info.maxDeg);//Pie4
      pwm3_Servo5.easeTo(pwm3_5Info.maxDeg);//Pie5
      pwm3_Servo6.easeTo(pwm3_6Info.maxDeg);//Pie6
      pwm3_Servo7.easeTo(pwm3_7Info.maxDeg);//Pie7

      pwm2_Servo8.easeTo(pwm2_8Info.maxDeg); //Lower Util
      pwm2_Servo9.easeTo(pwm3_9Info.maxDeg); //UpperUtil
      pwm2_Servo4.easeTo(pwm2_4Info.maxDeg); //Interface Door
      pwm2_Servo6.easeTo(pwm2_6Info.maxDeg); //Gripper Door
      pwm2_Servo11.easeTo(pwm2_11Info.maxDeg); //Buzz Door
      pwm1_Servo4.easeTo(pwm1_4Info.maxDeg);//DataPanel Door
      pwm2_Servo7.easeTo(pwm2_7Info.maxDeg); //CBI

Tim

ArminJo, 2019-11-21:
Did you set the start degree for all Servos? - with servo_xyz. write(<startDegree>)?

This is the code of easeTo(), it blocking waits for update() to return that movement has finished.
```
void ServoEasing::easeTo(int aDegree) {
    easeTo(aDegree, mSpeed);
}

void ServoEasing::easeTo(int aDegree, uint16_t aDegreesPerSecond) {
    startEaseTo(aDegree, aDegreesPerSecond, false);
    do {
        // First do the delay, then check for update, since we are likely called directly after start and there is nothing to move yet
        delay(REFRESH_INTERVAL / 1000); // 20 ms - REFRESH_INTERVAL is in Microseconds
    } while (!update());
}
```
So I cannot imagine how 3 Servos are moving at the same time except **speed is too high**, then software is faster than servos. Please check your speed settings.
In your video it looks like this could be the reason.

Eebel, 2019-11-25:
I start out the init with:
     pwm1_Servo4.attach(pwm1_4Info.pinNum, 1735, 2165);//DataPanel Door

     pwm2_Servo1.attach(pwm2_1Info.pinNum, 544, 2400, -100, 100);//Buzz Slide
    if(digitalRead(PIN_BUZZSLIDELIMIT) == HIGH){
        pwm2_Servo1.startEaseTo(100); //  Only move the Servo if the limit is not met and then only until limit so use non-blocking
    }
      
      pwm2_Servo4.attach(pwm2_4Info.pinNum, 1650, 2150);//Gripper Door
      pwm2_Servo6.attach(pwm2_6Info.pinNum, 1800, 2270);//Interface Door 2258
      pwm2_Servo6.setReverseOperation(true);
      pwm2_Servo7.attach(pwm2_7Info.pinNum, 1575, 2025);//CBI Door, 1980,1936
      pwm2_Servo7.setReverseOperation(true);
      pwm2_Servo8.attach(pwm2_8Info.pinNum, 480, 1848); //Lower UtiltyArm
      pwm2_Servo8.setReverseOperation(true);
      pwm2_Servo9.attach(pwm2_9Info.pinNum, 660, 2139);//Upper UtilityArm
      pwm2_Servo9.setReverseOperation(true);
      pwm2_Servo11.attach(pwm2_11Info.pinNum, 650,2305);//Buzz Door, 690,935-1797,2295
      pwm2_Servo11.setReverseOperation(true);
      pwm2_Servo12.attach(pwm2_12Info.pinNum, 700, 2200);//Gripper Lift
      pwm2_Servo13.attach(pwm2_13Info.pinNum, 2000, 2600);//Gripper
      pwm2_Servo13.setReverseOperation(true);
      pwm2_Servo14.attach(pwm2_14Info.pinNum, 700, 2200);//InterfaceLifter
      pwm2_Servo14.setReverseOperation(true);
      pwm2_Servo15.attach(pwm2_15Info.pinNum, 737, 1850);//Interface5

      pwm3_Servo1.attach(pwm3_1Info.pinNum,920,1600);//Pie 1
      pwm3_Servo1.setReverseOperation(true);
      pwm3_Servo2.attach(pwm3_2Info.pinNum, 1064,1660);//Pie 2
      pwm3_Servo2.setReverseOperation(true);
      pwm3_Servo3.attach(pwm3_3Info.pinNum, 980,1568);//Pie 3
      pwm3_Servo3.setReverseOperation(true);
      pwm3_Servo4.attach(pwm3_4Info.pinNum,873,1575);//Pie 4
      pwm3_Servo4.setReverseOperation(true);
      pwm3_Servo5.attach(pwm3_5Info.pinNum, 840,1625);//Pie 5
      pwm3_Servo5.setReverseOperation(true);
      pwm3_Servo6.attach(pwm3_6Info.pinNum, 1185,1855);//Pie 6
      pwm3_Servo6.setReverseOperation(true);
      pwm3_Servo7.attach(pwm3_7Info.pinNum, 608, 1260);//Pie 7
      pwm3_Servo7.setReverseOperation(true);

Then, I run close all to set the initial position.  I also use closeAll() when I need to close the panels.
      closeAll();

void closeAll() {
      pwm2_Servo8.write(pwm2_8Info.minDeg); //Lower Util
      delay(300);
      pwm2_Servo9.write(pwm2_9Info.minDeg); //UpperUtil
      delay(300);
      pwm3_Servo1.write(pwm3_1Info.minDeg);
      if (!lifeformActivated){
          pwm3_Servo2.write(pwm3_2Info.minDeg);
      }
      pwm3_Servo3.write(pwm3_3Info.minDeg); //Pie
      pwm3_Servo4.write(pwm3_4Info.minDeg);
      pwm3_Servo5.write(pwm3_5Info.minDeg);
      pwm3_Servo6.write(pwm3_6Info.minDeg);
      pwm3_Servo7.write(pwm3_7Info.minDeg);

      if (!interfaceActivated){
      pwm2_Servo4.write(pwm2_4Info.minDeg); //Interface Door
         //delay(200);
      }
      if (!gripperActivated){
        pwm2_Servo6.write(pwm2_6Info.minDeg); //Gripper Door
         //delay(200);
      }
      pwm2_Servo7.write(pwm2_7Info.minDeg); //CBI Door
      if (!buzzActivated){
        pwm2_Servo11.write(pwm2_11Info.minDeg); //Buzz Door
         //delay(200);
      }
      pwm1_Servo4.write(pwm2_4Info.minDeg);// DataPanel
}

So, each servo is initialized to its minimum value.  I lowered the speed to from 600 to 150 (which is too slow for this application).  I have each servo set to the min and max values using a PWM servo tester that lets me see the pulse value of the servo as a number.  Then each servo min and max values are 0 and 180 degrees.  To compensate for this I did up the servo speed.  However, since it is only moving 45 to 90 degrees, the servo is not moving too fast.  But as a test I set the  value to 150.  The result is the sameas before, but it happens slower.  It seems that after some time the library is not keeping track of the time and just executes whatever is in the queue.

https://youtu.be/YrgOmku7gM0

Tim

ArminJo, 2019-11-25:
Hi Tim,
if you want, you can send me the complete code and your modified ServoEasing.h to armin.arduino@gmail.com.
Your error looks like the value in ServoEasing.h 
`#define MAX_EASING_SERVOS 16 // One PCA9685 has 16 outputs. You must MODIFY this, if you have more than one PCA9685 attached!` 
is at least 4 to low.

Please check at least the last attach call.
```
    /*
     * Check at least the last call to attach()
     */
    if(pwm3_Servo7.attach(pwm3_7Info.pinNum, 608, 1260) == INVALID_SERVO) {
        Serial.println(F("Error attaching servo 3.7"));
        while (true) {
            digitalWrite(LED_BUILTIN, HIGH);
            delay(100);
            digitalWrite(LED_BUILTIN, LOW);
            delay(100);
        }
    }
```

Eebel, 2019-11-25:
You figured it out!  The #define MAX_EASING_SERVOS was set to 16 from the last update.  I had it set to 48 as I have three servo expanders.  However, I failed to reset the value after the last update.  Right now I am only using 20 servos across three expanders.  Should I set the MAX_EASING_SERVOS to 20 for performance or does it not matter as long as it's not lower than the number of servos?

Thank you again,

Tim

ArminJo, 2019-11-25:
Thank you too for helping to improve the library!

I built a new release with:
```
    Improved INVALID_SERVO handling.
    Speed 0 (not initialized) handling.
    Fixed bug in ThreeServos example.
```
In the readme there is now described [how to set global symbols for Sloeber IDE](https://github.com/ArminJo/ServoEasing#using-sloeber).

And the MAX_EASING_SERVOS size does not affect the performance, only the memory needed.

Thanks
Armin

closed 2019-11-25 by ArminJo