ArminJo/ServoEasing #15: ESP8266 with PCA9685 insane movements

PaddyCube, 2019-12-03:
Hi,

I'm building a 4-legged robot and use your library for smooth movement. It use an ESP8266 to get commands by Wifi.

At Ssetup, I attach all servos and use Servo.write to bring them to start position at 90 degrees. This work so far. But each time I send the very first ease command to any servo, all attached servos start to move insane around. This dance take some seconds before all servos stop. After this, I can control each servo correct one by one without trouble.
Because I'm affraid to fry my motors, I currently have only the four shoulder motors attached to PCA9685 pin 0 3 6 and 9. All others are not attached and not initialized with Servo.write

Any idea why the first command drive all servos crazy?

ArminJo, 2019-12-04:
Hi,
I have no idea, since on an AVR board it was tested with 3 expander boards and all worked well.
Maybe a bug in the I2C implementation?

Do you use the Arduino IDE?
Are you able to see output on serial Monitor? If yes, you can open the library file *ServoEasing.h* and comment out the line `#define TRACE` or `#define DEBUG`.
This will print internal information visible in the Arduino *Serial Monitor*.

If you send me the code snippets for the I2C and Servo initialization and Servo control to armin.arduino@gmail.com , I will try to reproduce it on my ESP8266 board the next days.

I hope we can fix the problem together.
Armin

PaddyCube, 2019-12-04:
Hello,

thanks for your fast response.
Yes, I'm using Arduino IDE and I'll give the debug trace a try.

More details by mail

ArminJo, 2019-12-04:
Thanks for the mail, the issue is fixed in the new 1.4.3 Version.
The ESP8266 startup code does not initialize the BSS segment with 0's and therefore some essential variables had random values.

closed 2019-12-04 by ArminJo