ArminJo/ServoEasing #17: I2C-PCA9685 detection failed /  isMovingAndCallYield() crashes

CRprog, 2020-03-08:
Your library is great and I use it on the ESP32 together with the PCA9685 (1x)
and up to 6 servos attached. First of all: The servos already working as expected.
(Arduino 1.8.10, lastest ESP-Pack, lastest ServoEasing library)

But I've found two minor problems in "PCA9685_Expander.ino" ... I hope that's not my fault:
(1) 
"Error: Communication with I2C was successful, but found no I2C device attached at address: 0x40"
Cause: Wire.endTransmission() returns always 1: data too long to fit in transmit buffer
But as I said above: The servos are correctly moving afterwards ...
(2)
OK:      `while (Servo2.isMoving()) { blinkLED(); }`
Crash:  `while (Servo1.isMovingAndCallYield()) {  }`            (Immediatly, if called)

Any ideas ?

ArminJo, 2020-03-08:
Thanks for the valuable feedback, I will test it and give feedback.

ArminJo, 2020-03-09:
You are totally right, I added wire.begin() in the PCA9685_Expander.ino example.
To (2). It does not crash for me, but the nonlinear movement does not work correctly, so I will look at it.

ArminJo, 2020-03-09:
OK, I removed the yield from all but the ESP8266 and published a new release.
Thanks again for the feedback!!!
Armin

CRprog, 2020-03-09:
Nothing to thank, you saved me many days to self-build a similar (but certainly more primitive) parallel servo control for PCA9685.
Regarding (2) and the update to 1.5.1: 
The behavior is now different, but there are still crash-problems with isMovingAndCallYield().
To simplify matters, I have attached a shortened test code (ESP32 only) with some comments. 
[EaseMinimisedCrashcode.ino.txt](https://github.com/ArminJo/ServoEasing/files/4309478/EaseMinimisedCrashcode.ino.txt)

ArminJo, 2020-03-10:
Thanks for the example, it helped me to find the error.
The cause is described at line 470 in ServoEasing.h.
Use the new function areInterruptsActive() instead of isMovingAndCallYield().
I changed all examples in the new release.

CRprog, 2020-03-10:
My tests are now also running without problems. Nice.
Don't cares: 
- areInterruptsActive() is missing in keywords.txt
- The fileversion differences (library.properties 1.5.1/ ServoEasing.h 1.5.2 for example) are for now desired, I suspect?
Anyway, I will close this issue tomorrow, all done. :-)
Maybe I must come back later (new "issue") for a question or 2-3 ideas/wishes. Best regards.

closed 2020-03-11 by CRprog