ArminJo/ServoEasing #18: Trying what I think is basic example and it does not seem to move the servo

vkbrihma, 2020-03-26:
Hey!

So I am trying to use the library to move the servo in a non-blocking manner.  I was able to get the blocking code of easeTo to work but when I changed to startEaseTo it did not seem to work.  Here is is the code I am testing.

`#include <Arduino.h>

#include "ServoEasing.h"

#define VERSION_EXAMPLE "1.4"

#define INFO // to see serial output of loop
const int SERVO1_PIN = 9;
const int SERVO1_MAX = 170;
const int SERVO1_MIN = 110;
#if !defined(LED_BUILTIN) && !defined(ESP32)
#define LED_BUILTIN PB1
#endif
ServoEasing Servo1;

void setup() {
    pinMode(LED_BUILTIN, OUTPUT);
    Serial.begin(115200);
  #if defined(__AVR_ATmega32U4__)
    while (!Serial); //delay for Leonardo, but this loops forever for Maple Serial
  #endif
  #if defined(SERIAL_USB)
    delay(2000); // To be able to connect Serial monitor after reset and before first printout
  #endif
    // Just to know which program is running on my Arduino
    Serial.println(F("START " __FILE__ "\r\nVersion " VERSION_EXAMPLE " from " __DATE__));

    // Attach servo to pin
    Serial.print(F("Attach servo at pin "));
    Serial.println(SERVO1_PIN);
    if (Servo1.attach(SERVO1_PIN) == INVALID_SERVO) {
        Serial.println(F("Error attaching servo"));
    }

    /**************************************************
     * Set servos to start position.
     * This is the position where the movement starts.
     *************************************************/
    Serial.print(F("before first position "));
    Servo1.write(SERVO1_MIN);
    Serial.print(F("after first position "));
    // Wait for servo to reach start position.
    delay(500);
}

void blinkLED() {
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
    digitalWrite(LED_BUILTIN, LOW);
    delay(100);
}

void loop() {
  // put your main code here, to run repeatedly:
  //Servo1.setSpeed(10);  // This speed is taken if no further speed argument is given.
  Servo1.setEasingType(EASE_CUBIC_IN_OUT);
  Serial.print(F("before first ease "));
  Servo1.startEaseTo(SERVO1_MAX, 50);
  while (areInterruptsActive()) {
          Serial.print(F("in easing "));
; // wait for servo to stop
    }
  
  Serial.print(F("after first ease "));
  delay(7000);
//  Servo1.startEaseTo(SERVO1_MIN, 50);
//  delay(7000);

}`

ArminJo, 2020-03-26:
Hi,
does the example OneServo run sucessfull for you?

vkbrihma, 2020-03-27:
It gets up to the 45 degree rotation and stops

ArminJo, 2020-03-27:
Which board, which pin, which IDE?

vkbrihma, 2020-03-29:
Pin 9 using the mkr-wifi-1010

ArminJo, 2020-03-29:
Please try the new version from the repository ang give feedback.

vkbrihma, 2020-03-31:
0 to 90 works then it stops

`Sketch uses 38364 bytes (14%) of program storage space. Maximum is 262144 bytes.
Atmel SMART device 0x10010005 found
Device       : ATSAMD21G18A
Chip ID      : 10010005
Version      : v2.0 [Arduino:XYZ] Mar 19 2018 09:45:14
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
done in 0.705 seconds

Write 38524 bytes to flash (602 pages)

[===                           ] 10% (64/602 pages)
[======                        ] 21% (128/602 pages)
[=========                     ] 31% (192/602 pages)
[============                  ] 42% (256/602 pages)
[===============               ] 53% (320/602 pages)
[===================           ] 63% (384/602 pages)
[======================        ] 74% (448/602 pages)
[=========================     ] 85% (512/602 pages)
[============================  ] 95% (576/602 pages)
[==============================] 100% (602/602 pages)
done in 0.204 seconds

Verify 38524 bytes of flash with checksum.
Verify successful
done in 0.036 seconds
CPU reset.`

ArminJo, 2020-04-03:
Hi,
if you want you can try to change 3 lines
`GCLK->CLKCTRL.reg = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK1 | GCLK_CLKCTRL_ID(GCM_TC4_TC5)); // GCLK1=32kHz,  GCLK0=48Mhz`
to
`GCLK->CLKCTRL.reg = (uint16_t) (GCLK_CLKCTRL_CLKEN | **GCLK_CLKCTRL_GEN_GCLK0** | GCLK_CLKCTRL_ID(GCM_TC4_TC5)); // GCLK1=32kHz,  GCLK0=48Mhz`
and

`TC5->COUNT16.CTRLA.reg |= TC_CTRLA_PRESCALER_DIV1 | TC_CTRLA_ENABLE; `
to
` TC5->COUNT16.CTRLA.reg |= TC_CTRLA_PRESCALER_DIV1024 | TC_CTRLA_ENABLE;`

and 
`TC5->COUNT16.CC[0].reg = (uint16_t) ((32768 / 50) - 1);`
to
 `TC5->COUNT16.CC[0].reg = (uint16_t) ((SystemCoreClock/1024) / 50 - 1);`

This changes the clock source.

vkbrihma, 2020-04-05:
Going to switch to an ESP32 and see if I can get it working there first

ArminJo, 2020-04-17:
I ordered a SAMD board, but until it arrives I am not able to test the SAMD version.

ArminJo, 2020-05-09:
Fixed 😀

closed 2020-05-09 by ArminJo