ArminJo/ServoEasing #19: One Servo example crashes on ESP32

veitk, 2020-03-29:
## Bug Report

### Arduino Platform
* [* ] ESP32 board.

### IDE
* [* ] Arduino IDE

### Example expressing the bug
* [* ] OneServo

### Pin(s) used for servo if not default
18

### Current behavior

content of the serial log:

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10312
load:0x40080400,len:6460
entry 0x400806a4
START C:\Users\veit\AppData\Local\Temp\arduino_modified_sketch_582483\OneServo.ino
Version 1.4 from Mar 29 2020
Attach servo at pin 18
Move to 90 degree with 10 degree per second blocking
Move to 180 degree with 30 degree per second using interrupts
Move to 45 degree in one second using interrupts
Move to 135 degree and back nonlinear in one second each using interrupts
CORRUPT HEAP: Bad head at 0x3ffb84b0. Expected 0xabba1234 got 0x3ffb9a18
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x400d598b on core 1

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400858a4:0x3ffb1e10 0x40085b19:0x3ffb1e30 0x400d598b:0x3ffb1e50 0x4008a26d:0x3ffb1e80 0x40083f96:0x3ffb1ea0 0x400843dd:0x3ffb1ec0 0x4000bec7:0x3ffb1ee0 0x400d6ce8:0x3ffb1f00 0x400d207b:0x3ffb1f20 0x400d1766:0x3ffb1f50 0x400d17db:0x3ffb1f70 0x400d10a7:0x3ffb1f90 0x400d2c59:0x3ffb1fb0 0x40089a75:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

### Expected behavior
run the example smoothly ;-)

### Additional context

veitk, 2020-03-29:
i just pulled the espressif/esp32 stuff and my log now looks like this:

Backtrace: 0x400d7040:0x3ffb7890 0x400d710f:0x3ffb78b0 0x40087fed:0x3ffb78d0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10044
load:0x40080400,len:5900
entry 0x400806ac
START C:\Users\veit\AppData\Local\Temp\arduino_modified_sketch_789998\OneServo.ino
Version 1.4 from Mar 29 2020
Attach servo at pin 18
Move to 90 degree with 10 degree per second blocking
Move to 180 degree with 30 degree per second using interrupts
Move to 45 degree in one second using interrupts
Move to 135 degree and back nonlinear in one second each using interrupts
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d7040  PS      : 0x00060533  A0      : 0x800d7112  A1      : 0x3ffb7890  
A2      : 0x3ffb84f0  A3      : 0x00000000  A4      : 0x010e7c94  A5      : 0x00000000  
A6      : 0x3ffaff9c  A7      : 0x3ffb7e54  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x3ffb84f0  A11     : 0x40082d38  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060320  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400d7040:0x3ffb7890 0x400d710f:0x3ffb78b0 0x40087fed:0x3ffb78d0

ArminJo, 2020-03-30:
Please use the oneServo example from the repo, it has changed.
In short do **not** use `isMovingAndCallYield()` please use `areInterruptsActive()`

The isMovingAndCallYield call is dangerous for ESP32, since the timer is detached AFTER mServoMoves is set to false in handleServoTimerInterrupt(), and one core may check mServoMoves and started a new move with initializing the timer, and then the timer gets detached by handleServoTimerInterrupt() which leads to an error: CORRUPT HEAP: Bad head at ...

ArminJo, 2020-04-17:
With the current version and using `areInterruptsActive()` it works!

closed 2020-04-17 by ArminJo