ArminJo/ServoEasing #20: Compiling error on ESP32

vkbrihma, 2020-04-05:
## Bug Report

### Arduino Platform
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP8266 board
* [X ] ESP32 board
* [ ] STM32 board
* [ ] Other - please specify

### IDE
* [X ] Arduino IDE
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [ ] PlatformIO IDE
* [ ] Other

### Example expressing the bug
* [ ] Simple
* [X ] OneServo
* [ ] TwoServos
* [ ] AsymmetricEasing
* [ ] PCA9685_Expander
* [ ] PCA9685_ExpanderFor32Servos
* [ ] ContinuousRotatingServo
* [ ] Other

### Pin(s) used for servo if not default

### Current behavior

In file included from C:\Users\vkbri\Documents\Arduino\OneServo\OneServo.ino:27:0:

C:\Users\vkbri\Documents\Arduino\libraries\ServoEasing-master\src/ServoEasing.h:79:26: fatal error: ESP32Servo.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

Tried to compile and it can't seem to find the other library.  I am able to run the examples here: https://github.com/jkb-git/ESP32Servo

### Expected behavior

Expected it to compile after I hadded the ESP32Servo libarary.  I am able to run the examples from the ESP32Servo libary fine.

### Additional context

ArminJo, 2020-04-06:
Hi,
do you see the lines
```
Alternatives for ESP32Servo.h: [ESP32Servo@0.7.0]
ResolveLibrary(ESP32Servo.h)
  -> candidates: [ESP32Servo@0.7.0]
```
in your verbose compile output?
If not, then you must install it with the library manager / Ctrl + Shift l

vkbrihma, 2020-04-06:
Nice that worked well!  Thanks for all your help!

closed 2020-04-06 by vkbrihma