ArminJo/ServoEasing #21: Regulary crashes with ESP32/PCA9685, but never with ESP8266/PCA9685

jibeer, 2020-04-11:
## Bug Report

### Arduino Platform
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP8266 board
* [X] ESP32 board
* [ ] STM32 board
* [ ] Other - please specify

### IDE
* [ ] Arduino IDE
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [ ] PlatformIO IDE
* [ ] Other

### Example expressing the bug
* [ ] Simple
* [ ] OneServo
* [ ] TwoServos
* [ ] AsymmetricEasing
* [ ] PCA9685_Expander
* [ ] PCA9685_ExpanderFor32Servos
* [ ] ContinuousRotatingServo
* [ ] Other

### Pin(s) used for servo if not default

### Current behavior

Regulary crashes with ESP32/PCA9685, but never with ESP8266/PCA9685 (maybe when under stress...)
4 servos in use.
ROCK SOLID with ESP8266/PCA9685 !

### Expected behavior

### Additional context

ArminJo, 2020-04-12:
Do you use the latest version?
What is the crash output?
Is one Servo moving before the crash?

ArminJo, 2020-04-17:
With the current version and using areInterruptsActive() it works!

closed 2020-04-17 by ArminJo

jibeer, 2020-04-17:
Hello. Sorry not to be able to help more. But seems that you found the bug,
congrats. Let me say again that your library is awesome! Respect.
All the best!