ArminJo/ServoEasing #22: Is it possible to use a PCA9685 AND Servos connected to Arduino?

Eebel, 2020-05-28:
## Bug Report
Is it possible to use PCA 9685 expander and also run another servo from the Mega 2560.

I'm building another droid.  I communicate with three servos in the head of the droid.  I have another servo in the body of the droid.  Space is extremely tight in this build.
### Arduino Platform
* [ ] Arduino ATmega328* board (UNO, Nano)
* [X ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP8266 board
* [ ] ESP32 board
* [ ] STM32 board
* [ ] Other - please specify

### IDE
* [X ] Arduino IDE
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [ ] PlatformIO IDE
* [ ] Other

### Example expressing the bug
* [ ] Simple
* [ ] OneServo
* [ ] TwoServos
* [ ] AsymmetricEasing
* [X ] PCA9685_Expander
* [ ] PCA9685_ExpanderFor32Servos
* [ ] ContinuousRotatingServo
* [ ] Other

### Pin(s) used for servo, if not default
46 from Mega 2560 and pins 5,6,7 on PCA Expander
### Version
* [ ] Yes I use the latest version
* [ X] Other - please specify
1.4.1. 

### Current behavior

// init code goes here.. includes and global variables and the like

const int pinArmServo = 46;   //Arm Angle is driven from 2560 due to location in the body
//------Rest of the servos are in the head and on the PCA 9685 Expander------
const int pinTiltServo = 5;     //Tilt Side-To-Side
const int pinNeckTurn = 6;   //Turn Neck
const int pinHeadNod = 7;   //Nod Up-And-Down

ServoEasing ArmServo; //Not going to the PCA9685
ServoEasing HeadTiltServo(PCA9685_DEFAULT_ADDRESS, &Wire);
ServoEasing NeckTurnServo(PCA9685_DEFAULT_ADDRESS, &Wire);
ServoEasing HeadNodServo(PCA9685_DEFAULT_ADDRESS, &Wire);
}

struct ServoControlStruct {
    uint16_t minDegree;
    uint16_t maxDegree;
    uint16_t center;
};

//-=-=-=-Structure for easier to read code-=-=-
ServoControlStruct ArmServoControl;
ServoControlStruct NeckTurnServoControl;
ServoControlStruct HeadNodServoControl;
ServoControlStruct HeadTiltServoControl;

//-=-=- Head servo targets-=-=- main loop will set these according to IMU and user inputs
uint8_t tNewArmPosition = 90;
uint8_t tNewHorizontal = 90;
uint8_t tNewVertical = 90;
uint8_t tNewTilt = 90;
uint16_t armCenter = 100;
uint16_t neckCenter = 90;
uint16_t nodCenter = 90;
uint16_t tiltCenter = 90;

void setup(){

  Serial.begin(115200);
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy etc
  if (Usb.Init() == -1) {
    Serial.print(F("\r\nOSC did not start"));
    while (1); //halt
  }
  Serial.print(F("\r\nPS3 Bluetooth Library Started"));
  setupServos();
}

void loop(){
////code to read PS3 inputs and grab Information to send to servos
// it sets values for tNewHorizontal, tNewArmPosition, tNewVertical, tNewTilt
//...

//The ArmServo doesn't move.  All the servos connected to the PCA expander work very smoothly with the PS3 control inputs.

  if (!ArmServo.isMoving()) {
        ArmServo.setEaseTo(tNewArmPosition);
        synchronizeAllServosAndStartInterrupt();
    }
     if (!NeckTurnServo.isMoving()) {
        NeckTurnServo.setEaseTo(tNewHorizontal);
        synchronizeAllServosAndStartInterrupt();
    }
    if (!HeadNodServo.isMoving()) {
       HeadNodServo.setEaseTo(tNewVertical);
        synchronizeAllServosAndStartInterrupt();
    } 
    if (!HeadTiltServo.isMoving()) {
        HeadTiltServo.setEaseTo(tNewTilt);
        synchronizeAllServosAndStartInterrupt();
    } 

}

void setupServos(){ 
    pinMode(pinArmServo, OUTPUT);
    ArmServo.attach(pinArmServo);
    HeadTiltServo.attach(pinTiltServo);
    NeckTurnServo.attach(pinNeckTurn);
    HeadNodServo.attach(pinHeadNod);
    
    ArmServoControl.minDegree = 70;
    ArmServoControl.maxDegree = 130;
    ArmServoControl.center = armCenter;
    NeckTurnServoControl.minDegree = 10;   
    NeckTurnServoControl.maxDegree = 170;  
    NeckTurnServoControl.center = neckCenter;
    HeadNodServoControl.minDegree = 50;     
    HeadNodServoControl.maxDegree = 140;   
    HeadNodServoControl.center = nodCenter;
    HeadTiltServoControl.minDegree = 35;  
    HeadTiltServoControl.maxDegree = 155; 
    HeadTiltServoControl.center = tiltCenter;

    //ArmServo.write(0);
    NeckTurnServo.write(NeckTurnServoControl.center);
    HeadNodServo.write(HeadNodServoControl.center);
    HeadTiltServo.write(HeadTiltServoControl.center);

    
    // Wait for servos to reach start position.
    delay(500);
}

### Expected behavior

Have all servos respond to PS3 control inputs
### Error output

None. No errors displayed.
### Additional context

ArminJo, 2020-05-29:
Hello Eebel,
to keep the library small, it is build with conditional compilation. 
Therefore it can either support the PCA expander or the Servo library, but not both.
It would need a rewrite of the libray to enable supporting both interfaces simultaneously 😢 
Sorry to tell you this.
Best regards
Armin

Eebel, 2020-05-29:
Thanks for the reply.  I was afraid that was the case.  I'll have to come up with a more traditional way to move that last servo.  ON a different note, I tried the latest version with my code and the servos moved very slowly.  I tried setting the set speed for all servos and it did not seem to help.  Did something change in the newer version that would I need to change to return the speeds of the servos?  Version 1.4.1 the servos move normally, the 1.6.1 version with the exact same code on my end makes the servos barely move.  Is my method of calling them in the main loop wrong?

ArminJo, 2020-05-29:
Very interesting.
Can you sent me the code to armin.arduino@gmail.com?
 I would like to test it.
There should be no difference between 1.4.1 and 1.6.1 this look like a bug at my side.

ArminJo, 2020-05-29:
The difference between 1.4.1 and 1.6.1 is that 1.6.1 initializes speed with 1 if not set to signal, that speed was not set. ( I changed this to 5 now).

With your request you triggered my ambition 😀 .
The current version in the repo is capable of doing this. See example PCA9685_ExpanderAndServo.

To control simultaneously servos with the Servo library e.g. which are directly connected to the board, comment out the line `#define USE_SERVO_LIB` in the library file *ServoEasing.h*.

ArminJo, 2020-05-29:
Hi Tim,
if it works for you (the regular and expander servos simultaneously) please give me feedback.
I will then publish the new release.
Thanks
Armin

Eebel, 2020-05-29:
OK, I'm on it.  I'll report back asap.

ArminJo, 2020-05-30:
Thanks, I just published the new release.

closed 2020-05-30 by ArminJo