ArminJo/ServoEasing #23: No timer for Teensy 4.1

Eebel, 2020-06-12:
## Bug Report

### Arduino Platform
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP8266 board
* [ ] ESP32 board
* [ ] STM32 board
* [x ] Other - please specify

### IDE
* [ ] Arduino IDE
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [ ] PlatformIO IDE
* [ x] Other

### Example expressing the bug
* [ ] Simple
* [ ] OneServo
* [ ] TwoServos
* [ ] AsymmetricEasing
* [x ] PCA9685_Expander
* [ ] PCA9685_ExpanderFor32Servos
* [ ] ContinuousRotatingServo
* [ ] Other

### Pin(s) used for servo, if not default

### Version
* [ x] Yes I use the latest version
* [ ] Other - please specify

### Current behavior

"No periodic timer support existent (or known) for this platform. Only blocking functions and simple example will run!"

### Expected behavior

### Error output

### Additional context

Eebel, 2020-06-12:
Using the Arduino IDE with the Teensyduino add-on.

ArminJo, 2020-06-13:
I added a timer, but I can not test it. Please test the latest version from repository.

Eebel, 2020-06-13:
Thanks, I’ll report.

Tim

Eebel, 2020-06-13:
It works very nicely.

Thank you for keeping the best servo library up-to-date!  I absolutely love the power of these tools you have created.

Thank you,

Tim

closed 2020-06-14 by ArminJo