ArminJo/ServoEasing #24: Detaching of the only servo

NereySer, 2020-06-20:
## Bug Report

### Arduino Platform
* [ ] All

### IDE
* [ ] Arduino IDE

### Version
* [ ] Yes I use the latest version

### Current behavior
```
#include "ServoEasing.h"
ServoEasing Servo1;
void setup () {
  Serial.begin(115200);
  delay(1000);
  
  Servo1.attach(D5);
  Servo1.detach();
  Servo1.attach(D5);
  Servo1.write(0);
  Servo1.startEaseTo(180);
}
void loop () {
}
```
This code causes an 28 exception ( LoadProhibitedCause )

### Expected behavior
Move servo to 180

### Error output

```
Decoding stack results
0x40201951: ServoEasing::update() at d:\��� ���㬥���\Arduino\libraries\ServoEasing-master\src\ServoEasing.cpp line 856
0x402019b4: updateAllServos() at d:\��� ���㬥���\Arduino\libraries\ServoEasing-master\src\ServoEasing.cpp line 1482
0x402019d8: handleServoTimerInterrupt() at d:\��� ���㬥���\Arduino\libraries\ServoEasing-master\src\ServoEasing.cpp line 1089
0x402042ba: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\sergs\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40204332: Ticker::_static_callback(void*) at C:\Users\sergS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\libraries\Ticker\src\Ticker.cpp line 71
0x4020250c: loop_task(ETSEvent*) at C:\Users\sergS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_main.cpp line 205
0x40202bbd: startWaveformClockCycles(uint8_t, uint32_t, uint32_t, uint32_t) at C:\Users\sergS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_waveform.cpp line 146
0x40201dee: Ticker::_attach_ms(unsigned int, bool, void (*)(void*), void*) at C:\Users\sergS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\libraries\Ticker\src\Ticker.cpp line 49
0x4020359d: uart_flush(uart_t*) at C:\Users\sergS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\uart.cpp line 544
0x401004e8: millis() at C:\Users\sergS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40201a5a: enableServoEasingInterrupt() at d:\��� ���㬥���\Arduino\libraries\ServoEasing-master\src\ServoEasing.cpp line 1224
0x40201ae0: ServoEasing::startEaseToD(int, unsigned int, bool) at d:\��� ���㬥���\Arduino\libraries\ServoEasing-master\src\ServoEasing.cpp line 707
0x40201b60: ServoEasing::startEaseTo(int, unsigned int, bool) at d:\��� ���㬥���\Arduino\libraries\ServoEasing-master\src\ServoEasing.cpp line 687
0x40100154: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\sergS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_main.cpp line 177
0x40100175: esp_schedule() at C:\Users\sergS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_main.cpp line 125
0x402026bd: loop_wrapper() at C:\Users\sergS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_main.cpp line 199
```

### Possible solving
Replace ServoEasing.cpp:393
```
            do {
                sServoArrayMaxIndex--;

                if (sServoArrayMaxIndex == 0) {
                    break;
                }
            } while (sServoArray[sServoArrayMaxIndex] == NULL);
```
with the
```
            while (sServoArrayMaxIndex > 0 && sServoArray[sServoArrayMaxIndex] == NULL) {
                sServoArrayMaxIndex--;
            }
```
or
```
            do {
                if (sServoArrayMaxIndex == 0) {
                    break;
                }

                sServoArrayMaxIndex--;
            } while (sServoArray[sServoArrayMaxIndex] == NULL);
```
if you wish

ArminJo, 2020-07-08:
Thanks!!!
Your patch is included since the 2.2.1 release.

closed 2020-07-08 by ArminJo

NereySer, 2020-07-13:
Sorry, it in 2.1.1 release
Thank you