ArminJo/ServoEasing #30: Use Microseconds instead of Degrees

euboty, 2020-11-23:
Hello ArminJo, thanks for your awesome work. I'm pretty new to all of this an was asking myself if there is a way to use the easing functions with microseconds instead of degrees?

Something like this:
_Servo1.startEaseToMicrosecond(1500);_

I did not found any examples/docs referring to that. Thanks for your time.

ArminJo, 2020-11-23:
**Servo speed** can be specified in **degree per second** or **milliseconds** for the complete move.
See [API](https://github.com/ArminJo/ServoEasing/blob/master/src/ServoEasing.h#L428).
Is this info sufficient for you?

euboty, 2020-11-23:
No that is not what I meant. I was thinking about the position in PWM. Just like this but for your easing functions:
_MyServo.writeMicroseconds()_ 
https://www.arduino.cc/en/Reference/ServoWriteMicroseconds

Problems is I've got all my positions for my servos in PWM Signal (Microseconds)  and not in degrees.

ArminJo, 2020-11-23:
How do you want to specify the speed?
InDegreesPerSecond or MicrosecondsPerSecond?
And the trim? 

For a first try. please change 
```
int ServoEasing::DegreeToMicrosecondsOrUnits(int aDegree) {
// For microseconds and PCA9685 units:
        return map(aDegree, 0, 180, mServo0DegreeMicrosecondsOrUnits, mServo180DegreeMicrosecondsOrUnits);
}
```
to 
```
int ServoEasing::DegreeToMicrosecondsOrUnits(int aDegree) {
// For microseconds and PCA9685 units:
    if (aDegree < 400) {
        return map(aDegree, 0, 180, mServo0DegreeMicrosecondsOrUnits, mServo180DegreeMicrosecondsOrUnits);
    } else {
        return aDegree;
    }
}
```

in ServoEasing.cpp.
Maybe this is the solution for you.

ArminJo, 2020-11-26:
Does it work for you?

euboty, 2020-11-29:
It works like a charm. Thank you so much!

closed 2020-11-29 by euboty