ArminJo/ServoEasing #33: Could you publish the most current version to PlatformIO?

wilbur4321, 2021-01-03:
I see it fixes the bug I just figured out, too, with negative directions (the int32_t cast that was needed) -- and I'd like to use that new parameter you just added, rather than degrees :)

Thanks so much!!

ArminJo, 2021-01-03:
Sorry, but I have no Platformio account and creating a new one failed, I got no mail :-(

ArminJo, 2021-01-04:
Now I hav an account but get the error
Error: ('Connection aborted.', ConnectionResetError(10054, 'Eine vorhandene Verbindung wurde vom Remotehost geschlossen', None, 10054, None)) 😢

ArminJo, 2021-01-04:
A new version of the package ServoEasing (2.3.3) was published at 2021-01-04 19:17:37 from 87.78.52.220.
The SHA-256 of this package was 6a9c3b928cce18fed2c1b790cf2608deaa7fcf990a9ac92b487f2c57487163e8.

closed 2021-01-04 by ArminJo

tobozo, 2021-11-04:
Hello @ArminJo,

I'm adding followup to this issue this on behalf of @mongonta0716 about a project that uses your library.

Looks like platformio registry has registered twice your library but missed the indexing of release 2.4.0.

Moreover, there are currently differences between [release 2.4.0](https://github.com/ArminJo/ServoEasing/releases/tag/v2.4.0) and the master branch where [library.properties](https://github.com/ArminJo/ServoEasing/blob/master/library.properties) claims to be `2.4.0` too.

Since platformio still randomly self-updates its registry by crawling the repositories, this creates an ambiguous situation where one library is based on the channel names, and the other is based on the github release tags.

[image]

This happened to me, and this is what I did: 

1) login to [platformio community](https://community.platformio.org/) site (using github auth) to : 
  - report the anomaly
  - have them merge all versions of the split library into a single instance
2) increment the current library version then run "pio package publish" from the root of the project
3) rule of thumb to avoid platformio skipping versions: increment the version number in the library.properties *every time* before pushing to the master

ArminJo, 2021-11-06:
Thanks for the information.
Seems they scan both library.json and library.properties. But they are both consistent, only different attributes :-(.
Is it better to remove the library.json? What is common sense?

>     2\. rule of thumb to avoid platformio skipping versions: increment the version number in the library.properties _every time_ before pushing to the master

Have you seen the master commit history?
What would you advise if I only want to use the master branch?

tobozo, 2021-11-06:
> What would you advise if I only want to use the master branch?

I usually increment the library version in the library.properties and library.json files, the run `pio package publish` from my PC *before pushing to the repository*.

This creates an archive from the local folder and submits it to platformio registry.

When the submission is complete (I get a confirmation email), I push my changes to the master, create a tag and produce a new github release so that both Arduino and platformio registries are in sync with the version numbers.

ArminJo, 2021-11-06:
Thanks!

> When a new release must be produced, I usually increment the library version in the library.properties and library.json files, then run `pio package publish` from my PC _before pushing to the repository_.

What is the advantage to have the increment only locally when running run pio package publish? Why not just after creating the release or is this possible too? And who may issue a `pio package publish` command?
Is there any documentaion of the way PlatformIo crawls the repos?
Thanks for your effort!

tobozo, 2021-11-06:
> And who may issue a pio package publish command?

only the library owner can do that, but since a token is used for the auth, I guess a github action could to that too

> Is there any documentaion of the way PlatformIo crawls the repos?

crawler frequency used to be on [this page](https://docs.platformio.org/en/latest/librarymanager/creating.html), but it's gone, maybe there's no crawler anymore ? 

[image]

ArminJo, 2021-11-06:
I am a bit lost. It is total unclear to me what the problem is, except of two choices for the same library, because they parse both files (why cant they fix that???) Isn't there any issue opened for that?
And no documentation is frankly no base for a good support.

tobozo, 2021-11-06:
> why cant they fix that???

Nobody reported the problem for your library, but [you are not alone](https://community.platformio.org/t/duplicate-library/2514).

> Isn't there any issue opened for that?

Not issue tracker that I know of, but creating a thread on the [community site](https://community.platformio.org/) using github auth works fine.

ArminJo, 2021-11-06:
> Not issue tracker that I know of, but creating a thread on the [community site](https://community.platformio.org/) using github auth works fine.

Can you create the thread on my behalf? I never used platformIO, I use Sloeber and Eclipse and sometimes Arduino IDE.

tobozo, 2021-11-06:
> Can you create the thread on my behalf?

I [did](https://community.platformio.org/t/duplicate-library-servoeasing/24372) but as a third-party I can't take decisions on your behalf. 

Also it won't publish the 2.4.0 and the next releases for you.

If you don't want to bother installing platformio locally, you can still create a github workflow for that, here's an [example](https://github.com/Foupi/Luos_NRF/blob/master/.github/workflows/pio_release.yml) using `workflow_dispatch`, should work as long as you increment the version before dispatching the workflow.

tobozo, 2021-11-06:
another reason why you should take ownership via github auth is to stop exposing your personal gmail address and have it replaced by the more anonymous github noreply version.

[image]

ArminJo, 2021-11-06:
> here's an [example](https://github.com/Foupi/Luos_NRF/blob/master/.github/workflows/pio_release.yml) using `workflow_dispatch`, should work as

Thanks, I will give it a try 😀 , Last question: where do I get the secrets.PIO_TOKEN?

tobozo, 2021-11-07:
To generate the token, you need to install platformio, there's an [Eclipse CDT](https://docs.platformio.org/en/latest/integration/ide/eclipse.html) but installing the CLI is enough for that.

- `pip install platformio`
- `pio account login` << your email as in the library.json (see [help page](https://docs.platformio.org/en/latest/core/userguide/account/cmd_login.html))
- `pio account token` <<< save the result (see [help page](https://docs.platformio.org/en/latest/core/userguide/account/cmd_token.html))

ArminJo, 2021-11-07:
Thanks
I opened an ["issue"](https://community.platformio.org/t/double-library-servoeasing/24375) and after resolving I can pulish the new version.

tobozo, 2021-11-07:
looks like the action is working, you can  bypass the yes/no prompt with `--non-interactive` as in `pio package publish --non-interactive`

ArminJo, 2021-11-07:
Success 🚀 
@tobozo Thanks for your patience 👍