ArminJo/ServoEasing #41: Linking error when library is used outside of main sketch (2.4.0)

Akhernar, 2022-01-09:
## Bug Report

### Version
* 2.4.0

### Current behavior
The switch to header-only library in release 2.4.0 completely broke compilation for me. 

Context: I use the library in my own class "MyClass". "MyClass" is defined in file MyClass.h, which includes ServoEasing.hpp. Member function definition is in MyClass.cpp. MyClass.h is included in the main sketch, where MyClass is used. 

The problem: After updating to 2.4.0 and changing `#include "ServoEasing.h"` to `#include "ServoEasing.hpp"` in MyClass.h my project compiles just fine, but then I get a ton of the following linker errors (only one error pasted here as an example, the same is repeated for all functions in ServoEasing.hpp, both free functions and ServoEasing class member functions):

```
.pio\build\uno\src\main.cpp.o (symbol from plugin): In function `QuadraticEaseIn(float)':
(.text+0x0): multiple definition of `QuadraticEaseIn(float)'
.pio\build\uno\src\MyClass.cpp.o (symbol from plugin):(.text+0x0): first defined here
```

This is due to the [One Definiton Rule](https://en.wikipedia.org/wiki/One_Definition_Rule), which is a common issue for header-only libraries. The errors happen because the library code is included separately in two different translation units (directly in MyClass.cpp, and through MyClass.h in main.cpp), which results in the multiple definition error during linking.

### Expected behavior
Library should compile and link correctly when used in custom files. For the moment I will just use 2.3.4, but I only see two possible long term solutions here:
* Revert back from *.hpp to  to *.cpp files (I'm actually curious here, what was the reason for the switch to a header-only library?)
* Make all ServoEasing.hpp functions `inline`, as suggested [here](https://stackoverflow.com/questions/3973218/header-only-libraries-and-multiple-definition-errors)

ArminJo, 2022-01-09:
Thanks for reporting 🥇 
I now added a chapter to the readme which should solve your problem.

# Using the new *.hpp files
In order to support [compile options](#compile-options--macros-for-this-library) more easily, the line `#include <ServoEasing.h>` must be changed to  `#include <ServoEasing.hpp>`, but only in your **main program (.ino file)**, like it is done in the examples.<br/>
In **all other files** you must use `#include <ServoEasing.h>`, otherwise you will get tons of **"multiple definition"** errors.

closed 2022-01-11 by ArminJo

Akhernar, 2022-01-12:
Hi, thanks for the quick response and sorry for the late reply.
Yes this works too, however it still feels strange to me that you have to include a file in the main when you are not actually using the library there. I understand this is to let the user define any compile option directly in the main, instead of adding them to the compiler command line.
Anyway I think you should also add to the readme a note to set compile option `SUPPRESS_HPP_WARNING` when using the library like this, otherwise you get a warning when in fact you are using the library as intended.

ArminJo, 2022-01-12:
Thanks a lot for the suggestion 👍 
I changed the warning
#warning You probably must change the line #include "ServoEasing.h" to #include "ServoEasing.hpp" in your ino file
to 
#warning You probably must change the line #include "ServoEasing.h" to #include "ServoEasing.hpp" in your ino file or define SUPPRESS_HPP_WARNING before the include to suppress this warning.