ArminJo/ServoEasing #43: Interrupt startEaseTo functions not working on Arduino Nano Every

Coremination, 2022-01-28:
## Bug Report

### Board
* [ ] Arduino megaAVR board (NanoEvery)

### IDE
* [ ] Arduino IDE

### Example to reproduce the issue
* [ ] OneServo

### Version
* [ ] Yes I use the [latest repo version](https://github.com/ArminJo/ServoEasing/archive/master.zip) and verified this!

### Current behavior
With Arduino Nano Every all of the startEaseTo functions that use interrupts are non functional. Using the examples and code within other applications it doesn't move the servo. Using the easeTo function (blocking) works well. 

I believe there might be something to do with using the Millis() TCA0 timer, it is referenced that the Every uses the TCB3 but I noticed that once the code had gone to use the startEaseTo function and changed the mode of the TCA0 timer the delay speed changed indicating it had affected the millis() counter.

ArminJo, 2022-01-28:
If I compile, I see:
```
'Starting C++ compile'
"... \arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections ... -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR   -DMILLIS_USE_TIMERB3
```
explicitly stating `-DMILLIS_USE_TIMERB3`
Can you check this on your Arduino IDE?

Coremination, 2022-01-29:
Interesting, you are right my compiler is also showing B3 as the millis timer:
```
..arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3
```

Not sure if it is helpful but did read this forum entry where a person modifies some of the properties of TCA0 and it increased the speed of the millis counter - [https://forum.arduino.cc/t/use-of-timer-counters/677895](url)

Even with the clock running at this faster speed, moving the servo using the interrupts isn't doing anything (if all that was happening was faster than logically would expect it to move but too fast..?)

ArminJo, 2022-01-29:
>  https://forum.arduino.cc/t/use-of-timer-counters/677895

The link is broken.
Maybe it is a good idea  if you (I have no Nano Every) open an issue [here](https://github.com/arduino/ArduinoCore-avr)

Coremination, 2022-01-29:
Sorry for that (works for me and tried it again).

I found this piece of code and tested on an Arduino Nano Every and it works (e.g. it picks up the interrupt set on TCA0), afraid I am not as deep an expert on these things but what it is doing might mean more to you in reference to how things are functioning in ServoEasing.hpp?

```
// =========================================================
// There is one type A timer/counter (TCA0) and it has three
// compare channels (CMP0, CMP1, CMP2).
//
// The clock prescaler (bits 3:1 of CTRLA register) is set to
// 64 by default and changing it breaks delay() and
// millis() (and maybe other things, not checked):
// increasing/decreasing it increases/decreases the length
// of "a second" proportionally. It also changes the PWM
// frequency of pins 5, 9, and 10, although they will still
// do PWM (increasing/decreasing the prescaler value
// decreases/increases the frequency proportionally).
//
// The period value (PER register) is set to 255 by default.
// Changing it affects the duty cycle of PWM pins (i.e. the
// range 0-255 in analogWrite no longer corresponds to a
// 0-100% duty cycle).
//
// PWM pin 5 uses compare channel 2 (CMP2), pin 9 channel
// 0 (CMP0), and pin 10 channel 1 (CMP1).
//
// If you don't mind losing PWM capability (or at least
// changing the frequency and messing up the duty cycle)
// on pins 5, 9, or 10, you can set up these channels for
// your own use.
//
// If you don't mind messing up delay() and millis(), you
// can change the prescaler value to suit your needs. In
// principle, since the prescaler affects millis() and
// delay() in an inversely proportional manner, this could
// be compensated for in your sketch, but this is getting
// messy.
//
// If the prescaler is not changed, the lowest interrupt
// frequency obtainable (by setting PER and CMPn to 65535)
// is about 3.8 Hz.
//
// The interrupt frequency (Hz) given a prescaler value
// and PER value is:
// f = 16000000 / (PER + 1) / prescaler
//
// The PER value for a required frequency f is:
// PER = (16000000 / prescaler / f) - 1
//
// The prescaler can take a setting of 1, 2, 4, 8,
// 16, 64, 256, or 1024. See the ATmega 4809 datasheet
// for the corresponding bit patterns in CTRLA.
//
// http://ww1.microchip.com/downloads/en/DeviceDoc/megaAVR-0-series-Family-Data-Sheet-40002015C.pdf
//
// The program below demonstrates TCA0 interrupts on
// compare channel 1, with and without changing the
// prescaler (change the value of CHANGE_PRESCALER
// from 0 to 1).
//
// PER is the TOP value for the counter. When it
// reaches this value it is reset back to 0. The
// CMP1 value is set to the same value, so that an
// interrupt is generated at the same time that the
// timer is reset back to 0. If you use more than
// one compare channels at the same time, all CMPn
// registers must have a value smaller or equal to PER or
// else they will not generate an interrupt. The
// smallest obtainable frequency is governed by the
// value of PER.
[... 56 lines cut ...]
    while(true) {
        if (i) {
            i = false;
            Serial.println("Interrupt!");
        }
        if (millis() - t >= 1000) {
            Serial.print("Still alive ");
            Serial.println(alive_count++);
            t = millis();
        }
    }
}

ISR(TCA0_CMP1_vect) {
    cli();
                
     Serial.println("Int");
    i = true;
    TCA0.SINGLE.INTFLAGS |= bit(5);                 // Clears the interrupt flag. Rather confusingly,
                                                    // this is done by setting a bit in the register
                                                    // to 1.
                                                    // Use bit(4) for CMP0, bit(5) CMP1, bit(6) CMP2
    sei();
}
```

Coremination, 2022-01-29:
I have fettled with the ServoEasing.hpp code and added the sections of code from above in to it:

```
#  elif defined(__AVR_ATmega4809__) // Uno WiFi Rev 2, Nano Every
    // TCB1 is used by Tone()
    // TCB2 is used by Servo, but we cannot hijack the ISR, so we must use a dedicated timer for the 20 ms interrupt
    // TCB3 is used by millis()
    // Must use TCA0, since TCBx have only prescaler %2. Use single mode, because it seems to be easier :-)
    #define CHANGE_PRESCALER    1
    unsigned int per_value;
    cli();
#   if CHANGE_PRESCALER == 0                        // Use default 64 prescale factor
        per_value = 0xFFFF;                         // Use maximum possible PER/CMP value (65535)
        TCA0.SINGLE.PER = per_value;                // Set period register
        TCA0.SINGLE.CMP1 = per_value;               // Set compare channel match value
        TCA0.SINGLE.INTCTRL |= bit(5);              // Enable channel 1 compare match interrupt.
                                                    // Use bit(4) for CMP0, bit(5) CMP1, bit(6) CMP2
#   elif CHANGE_PRESCALER == 1                      // Change prescale factor
        TCA0.SINGLE.CTRLA = B00000111;              // Prescaler set to 256. Use the following for
                                                    // other prescalers:
                                                    // B00000001    1
                                                    // B00000011    2
                                                    // B00000101    4
                                                    // B00000111    8
                                                    // B00001001    16
                                                    // B00001011    64
                                                    // B00001101    256
                                                    // B00001111    1024
        per_value = (((F_CPU / 1000000) * REFRESH_INTERVAL_MICROS) / 8);                         // Value required for 1 Hz (62499)
        TCA0.SINGLE.PER = per_value;                // Set period register
        TCA0.SINGLE.CMP1 = per_value;               // Set compare channel match value
        TCA0.SINGLE.INTCTRL |= bit(5);              // Enable channel 1 compare match interrupt.
                                                    // Use bit(4) for CMP0, bit(5) CMP1, bit(6) CMP2
#   endif
    sei();

    //old code
    //TCA0.SINGLE.CTRLB = TCA_SINGLE_WGMODE_NORMAL_gc;                        // Frequency mode, top = PER
    //TCA0.SINGLE.PER = (((F_CPU / 1000000) * REFRESH_INTERVAL_MICROS) / 8); // (F_CPU / 1000000) = clockCyclesPerMicrosecond()
//    TCA0.SINGLE.PER = ((clockCyclesPerMicrosecond() * REFRESH_INTERVAL_MICROS) / 8); // clockCyclesPerMicrosecond() is no macro here!
    //TCA0.SINGLE.INTFLAGS = TCA_SINGLE_OVF_bm;                               // reset interrupt flags
    //TCA0.SINGLE.INTCTRL = TCA_SINGLE_OVF_bm;                                // Overflow interrupt
    //TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL_DIV8_gc | TCA_SINGLE_ENABLE_bm;   // set prescaler to 8
```

and 

```
#elif defined(__AVR_ATmega4809__) // Uno WiFi Rev 2, Nano Every
ISR(TCA0_CMP1_vect) {
        cli();
    // Not tested, but with the experience, I made with the ATtiny3217, I guess it is required
    TCA0.SINGLE.INTFLAGS |= bit(5);                 // Clears the interrupt flag. Rather confusingly,
                                                    // this is done by setting a bit in the register
                                                    // to 1.
                                                    // Use bit(4) for CMP0, bit(5) CMP1, bit(6) CMP2

    sei();
        handleServoTimerInterrupt();
}
```

That has made it work now, I haven't had time to check if it does things to Millis() etc yet but hopefully you can maybe do a better job than my rubbish attempt :-)

ArminJo, 2022-01-30:
Thanks, I changed my code alsi to Compare1 interrupt.
Can you do me a favor and test the latest version of ServoEasing.hpp from the repository?

Coremination, 2022-01-31:
Pulled version 2.4.2 and trialled with an Arduino Nano Every and a single servo (Pin9) on my code and this version is non functional sorry.

ArminJo, 2022-02-01:
Thanks for testing 🥇 
I checked it with my ATtiny3217 and made some changes.
Can you test again?

Coremination, 2022-02-05:
Apologies for the delay - retested and still non functional.

Double checking by switching to my code as well to ensure it works / not working

ArminJo, 2022-02-05:
Thanks for testing.
Now I have no clue 😞 
So I have to wait for an Nano Every to appear for testing.

Coremination, 2022-02-05:
Sure it's not the SLI, CLI calls that I put in to the code I trialled?

ArminJo, 2022-02-05:
Quite sure!

Coremination, 2022-02-05:
Ok, One thing I didn't do is trial this latest it with the simple servo example, I used my main code so maybe something else is code is interfering. I will try it with the simple example when I can to see. Anything else I could be doing that might not be allowing it to work properly (when it might be ok)?

ArminJo, 2022-02-05:
I carefull tested it with a 3217 I have at hand. 
I bet you can find the error if you change your code successively to mine and report at which step it fails! Change the Interrupt type from your CMP to my OVF at last.

closed 2022-04-05 by ArminJo