ArminJo/ServoEasing #49: Remove power from servo

Eebel, 2022-05-04:
**Issues are expected to lead to changes in the repository, like code or documentation improvements or bug fixes.**<br/>
If you have handling problems or questions, consider to **open a [discussion](https://github.com/ArminJo/ServoEasing/discussions)** instead of an issue.
I am trying to remove all signals to a servo after I move it.  In Servo Easing 1.4.1, I used the .detach() function and it seemed to depower the servos until I moved them again.  In current version the servos always buzz when I am not trying to move them.  I am using three PCA 9685 servo expanders with an Arduino Mega 2560.
## Bug Report

### Board
* [X ] Arduino ATmega2560 board (Mega)

### IDE
* [ X] Arduino IDE

### Example to reproduce the issue

### Version
* [ X] Yes I use the [latest repo version](https://github.com/ArminJo/ServoEasing/archive/master.zip) and verified this!

Please delete all unchecked lines above :-)

### Pin(s) used for servo, if not default

### Current behavior

### Expected behavior

### Additional context

ArminJo, 2022-05-04:
Hi Eebel,
the code for detach is `setPWM(0); // set signal fully off`, which should disable any signal for this servo.
Can you check this with your Hardware?

Eebel, 2022-05-04:
I did a quick test by replacing the .detach() with .setPWM(0):  It still buzzed.

I have an event with my R2-D2 today. R2-D2 uses a totally different control system.

The droid that's having troubles is R5-D4. When I get back from that event,  I'll do some more testing with some servos not installed in R5-D4 and see if I can isolate the problem.

Thanks,

Tim

Eebel, 2022-05-05:
So, I setup a test rig with just servos and another Mega 2560 outside of my droid.  When I run the PCA9685_Expandere example sketch I don't hear any buzzing of the servo when it stops.  However, If I use my droid's sketch, it buzzes.  So, that leads me to believe I am doing something wrong.

I had my droid sketch working perfectly using the older library.  I had the uSec for limits of each servo.  I initially tried to use the ENABLE_MICROS_AS_DEGREE_PARAMETER set to still use the limits I had for each servo.  When that didn't work, disabled that macro and  I tried converting the uSec values to degrees using an interpolation formula.

Maybe you can tell me where I am going wrong.  I'll post some snippets of the code for a single servo.

Initially I do this:
#include <PS3BT.h>  //PS3 Bluetooth Files
#include <SPP.h>
#include <usbhub.h>
#include <Sabertooth.h>
#include <Servo.h>
#include <Wire.h>
//#include <Adafruit_PWMServoDriver.h> // I think I don't need this anymore due to ServoEasing library
//#include <Adafruit_Soundboard.h>

#include <NeoPatterns.h>
#include <MP3Trigger.h>
/*
 * !!! Uncomment use PCA9865 in ServoEasing.h to make the expander work !!!
 * Otherwise you will see errors like: "PCA9685_Expander:44:46: error: 'Wire' was not declared in this scope"
 */
#define USE_PCA9685_SERVO_EXPANDER
#define MAX_EASING_SERVOS 30
//#define ENABLE_MICROS_AS_DEGREE_PARAMETER
#define DEBUG

#include "ServoEasing.hpp"

ServoEasing pwm1_Servo4(0x40, &Wire); //Data Panel Door

In Setup() I run the following code.  I have disabled all but the DataPanelDoor servo for troubleshooting:

    pwm1_4Info.servoName = "DataPanel";
    pwm1_4Info.minDeg = 130;//1735;//0,65,77, 47
    pwm1_4Info.maxDeg = 157;//2165;//180, 37
    pwm1_4Info.pinNum = 4;
    pwm1_4Info.pwmNum = 1;

    pwm1_Servo4.attach(pwm1_4Info.pinNum,pwm1_4Info.minDeg ,pwm1_4Info.minDeg, pwm1_4Info.maxDeg);//DataPanel Door

    setSpeedForAllServos(600);//300,make servos fast. Go double since all servos think they are going 180 degrees but they are really only going 90 or less

In the Loop(), I read a PS3 controller and execute commands to move servos based on the commands from the controller.

Here is an example of a function that gets executed based on a PS3 controller command:

void closeAll(){
      pwm2_Servo8.write(pwm2_8Info.minDeg); //Lower Util
      delay(300);
      pwm2_Servo9.write(pwm2_9Info.minDeg); //UpperUtil
      delay(300);
      pwm3_Servo1.write(pwm3_1Info.minDeg);
      if (!lifeformActivated){
          pwm3_Servo2.write(pwm3_2Info.minDeg);
      }
      pwm3_Servo3.write(pwm3_3Info.minDeg); //Pie
      pwm3_Servo4.write(pwm3_4Info.minDeg);
      pwm3_Servo5.write(pwm3_5Info.minDeg);
      pwm3_Servo6.write(pwm3_6Info.minDeg);
      pwm3_Servo7.write(pwm3_7Info.minDeg);

      if (!interfaceActivated){
      pwm2_Servo4.write(pwm2_4Info.minDeg); //Interface Door
         delay(200);
      }
      if (!gripperActivated){
        pwm2_Servo6.write(pwm2_6Info.minDeg); //Gripper Door
         delay(200);
      }
      pwm2_Servo7.write(pwm2_7Info.minDeg); //CBI Door
      if (!buzzActivated){
        pwm2_Servo11.write(pwm2_11Info.minDeg); //Buzz Door
         delay(200);
      }
      pwm1_Servo4.write(pwm2_4Info.minDeg);// DataPanel
      delay(500);
      silenceServos();
      delay(400);

    silenceServos();
}

void silenceServos(){
    /////////////////////////////////////////////////
      //TODO:  stopAllServos(); does this work?
   //stopAllServos();
      pwm3_Servo1.detach(); //Pie1
      pwm3_Servo2.detach(); //Pie2
      pwm3_Servo3.detach(); //Pie3
      pwm3_Servo4.detach(); //Pie4
      pwm3_Servo5.detach(); //Pie5
      pwm3_Servo6.detach(); //Pie6
      pwm3_Servo7.detach(); //Pie7

      pwm2_Servo1.detach(); //BuzzSlide
      pwm2_Servo4.detach(); //LeftBreadPan
      pwm2_Servo6.detach(); //RightBreadPan
      pwm2_Servo7.detach(); //CBI Door
      pwm2_Servo8.detach(); //LowerUtil
      pwm2_Servo9.detach(); //UpperUtil
      pwm2_Servo11.detach(); //Buzz Door
      pwm2_Servo12.detach();  //GripperLift
      pwm2_Servo13.detach(); //Gripper
      pwm2_Servo14.detach(); //Interface Lift
      pwm2_Servo15.detach(); //Interface

      //pwm1_Servo4.detach();  //DataPanel Door
      pwm1_Servo4.setPWM(0);//All other servos are not active as they are never attached for troublshooting
      ////////////////////////////////////////////////
}

ArminJo, 2022-05-05:
OK, I see I may have some bugs in the ENABLE_MICROS_AS_DEGREE_PARAMETER implementation. I will revisit it and change it to the default behavior.
And I see a mistake in your code.#
the  `pwm1_Servo4.attach(pwm1_4Info.pinNum,pwm1_4Info.minDeg ,pwm1_4Info.minDeg, pwm1_4Info.maxDeg);`
has the signature of
`int aPin, int aInitialDegreeOrMicrosecond, int aMicrosecondsForServo0Degree, int aMicrosecondsForServo180Degree`
so you must use microseconds for 3. and 4. parameter.

Eebel, 2022-05-05:
I originally had the aMicrosecondsForServo0Degree set to the minimum servo value (i.e. 1735 uSec) and the aMicrosecondsForServo180Degree set to the max servo travel (i.e. 2035 uSec).  When that didn't work, I tried the values I posted above.

Should I be targeting the actual 0 and 180 values here?

Thanks,

Tim

closed 2022-05-30 by ArminJo