ArminJo/ServoEasing #52: Help with Bounce in my Code

ajkochev, 2022-06-18:
This isn't an issue really.  I would like to use your bounce function in my code to replace my bounceRun function.  I looked through the readme and opened a few examples but could not find one that uses the bounce.   Would it be possible to add an example that shows just how bounce works?

Below is my code.  Everything works good just would like a little bounce when the movement finishes and what I wrote isn't very good.

## Bug Report

### Board
* [ ] Arduino ATmega328* board (UNO, Nano)

### IDE
* [ ] Arduino IDE

### Example to reproduce the issue
* [ ] PCA9685_Expander

### Version
* [ ] Yes I use the [latest repo version](https://github.com/ArminJo/ServoEasing/archive/master.zip) and verified this!

Here is the code I have.  Everything works great.  Just want to replace my bounceRun function with something better.

``
// C/MRI Model Railroad Semaphore Signal Controller for JMRI.

#include <Wire.h>
#include <Adafruit_PWMServoDriver.h>
#include <CMRI.h>
#include <Auto485.h>
#include <Servo.h>
#include <ServoEasing.h>

#define CMRI_ADDR 3  //CMRI Node address.  Change if needed.
#define DE_PIN    2  //Ardino pin (TX enable ?)
#define N_POS   3   //Three positions per servo.
#define N_SERVO 16  //number of servos possible on a single PCA9685 board.

Adafruit_PWMServoDriver pwm1 = Adafruit_PWMServoDriver (0x40);
Auto485 bus (DE_PIN);
CMRI cmri (CMRI_ADDR, 24, 48, bus);  // addr, # inp,  # out and bus speed
Servo Current_Servo;

// ------------------------------------------------

int currentServo;

/*
Below sets servoSteps array for each Servo to move at a time.  Increase for faster movement.
Set in multiples of 5 only.
First number is for servo on port 0 second for servo on port 1, etc.  */
int servoSteps [N_SERVO] = {5,5,5,5, 5,5,5,5, 5,5,5,5, 5,5,5,5};

/*
Below servoPeriod array sets the delay time between when the servo is moved using servoSteps.  
This is an additional way to slow down servo movement even further than if servoSteps was at 1.  
Increase values for slower movement.
First number is for servo on port 0 second for servo on port 1, etc.  Number is in miliseconds.*/
int servoPeriod[N_SERVO] = {30,30,30,30, 30,30,30,30, 30,30,30,30, 30,30,30,30};

/*
Below servoVal array sets the three position values used to set each servos positions.  
First three are for servo on port 0 second three for port 1, etc.
Set in multiples of 5 only.
Values must go from least to highest for the code to work properly this cannot be changed. 
Be sure the servo and armature are oriented properly to account for this before mounting on your layout.
To get values needed, run a calibration sketch on your servos before implimenting this sketch
To make a semaphore a two color/two position, put the same values in two of the three values.  
See port 8 below. */

enum { Min, Mid, Max };

int servoVal [N_SERVO][N_POS] = {
[... 199 lines cut ...]
      pwm1.writeMicroseconds(curServo, m);
      delay(bouncePeriod[curServo]);
      m = m - bounceStep[curServo];
      pwm1.writeMicroseconds(curServo, m);
      //Move three steps forward
      delay(bouncePeriod[curServo]);
      m = m + bounceStep[curServo];
      pwm1.writeMicroseconds(curServo, m);
      delay(bouncePeriod[curServo]);
      m = m + bounceStep[curServo];
      pwm1.writeMicroseconds(curServo, m);
      delay(bouncePeriod[curServo]);
      m = m + bounceStep[curServo];
      //Move two steps back
      pwm1.writeMicroseconds(curServo, m);
      delay(bouncePeriod[curServo]);
      m = m - bounceStep[curServo];
      pwm1.writeMicroseconds(curServo, m);
      delay(bouncePeriod[curServo]);
      m = m - bounceStep[curServo];
      pwm1.writeMicroseconds(curServo, m);
    }
    //Set final position to what is was at start of function
    pwm1.writeMicroseconds(curServo, f);
}

ArminJo, 2022-06-19:
Just change easing type to Bounce like `Servo3.setEasingType(EASE_BOUNCE_OUT);` in the AsymmetricEasing example.

closed 2022-06-19 by ArminJo

ajkochev, 2022-06-20:
This is no help.  And I did look more into your code trying to make sense of it.   I still don’t understand how to put your code into mine when I need to move call my servos like this on a PCA9685 board:

 

pwm1.writeMicroseconds(curServo, m);

 

Closing the issue without asking if I was helped was rude as well.

ArminJo, 2022-06-20:
Please use another library!
And you now what an issue is??? in contrast to an discussion???