ArminJo/ServoEasing #85: setDegreeForAllServos parses parameters (degrees) as int, not float

navdotnetreqs, 2023-03-05:
## Bug Report

Hi and thanks for the library!

It seems that in ServoEasing.hpp, line 2300, void setDegreeForAllServos(uint_fast8_t aNumberOfServos, va_list *aDegreeValues)
parameters are converted to type int, even though ServoEasingNextPositionArray is float. Am I correct?

I think this is what causes my servos to fly all over the place if I use setDegreeForAllServos. If I assign the ServoEasingNextPositionArray-elements one at a time directly, everything works as expected.

### Board
* [ ] ESP32 board

### IDE
* [ ] PlatformIO IDE

navdotnetreqs, 2023-03-05:
Maybe this is relevant? https://stackoverflow.com/questions/11270588/variadic-function-va-arg-doesnt-work-with-float

ArminJo, 2023-03-05:
Thanks for reporting. I fixed it in the new 3.2.1 release by adding function setFloatDegreeForAllServos().

closed 2023-03-05 by ArminJo