ArminJo/ServoEasing #99: Does not respond to input

Leeeeerich, 2025-06-04:
Hello! Why can my servos stop responding? The thing is that after a while they don't respond to changes in input data, I thought it might be something like sleep, and added: `#define DISABLE_PAUSE_RESUME disabled`, it seemed to fix the situation, and it doesn't turn off, but in the first five minutes, then with incomprehensible time intervals, they still turn off.

My board: Blue Pill STM32F103C8T6 - China - ck.......
+ I have RS485, and I had problems before, but it work now.

```
#include <ServoEasing.hpp>
...
ServoEasing horServo;
ServoEasing verServo;
...
horServo.attach(HOR_SERVO_PIN, convertTo2500_500(512));
verServo.attach(VER_SERVO_PIN, convertTo2500_500(getVerticalWithin()));
...
horServo.startEaseTo(convertTo2500_500(horRawAngle), 60, START_UPDATE_BY_INTERRUPT);
verServo.startEaseTo(convertTo2500_500(getVerticalWithin()), 60, START_UPDATE_BY_INTERRUPT);
```

Leeeeerich, 2025-06-04:
I added a crutch:
`if (crutchDisablePause) {
    horServo.resumeWithInterrupts();
    footPause = false;
  }`

and it worked, but what's interesting is that I only sent it for one servo, for one object, the second one worked too.

ArminJo, 2025-06-04:
Sorry, your Example gives error:
`Compilation error: Configs.h: No such file or directory`
and I do not understand your questions.

Leeeeerich, 2025-06-04:
My config: 

`#define MIN_HOR_POS 0
#define MAX_HOR_POS 1023

#define MIN_VARTICAL_ANGLE 542 // 397
#define MAX_VERTICAL_ANGLE 755

#define MINIMUM_PULSE_WIDTH 400
#define DISABLE_PAUSE_RESUME disabled`

ArminJo, 2025-06-04:
Super, then all questions are answered!

closed 2025-06-04 by ArminJo

Leeeeerich, 2025-06-04:
> and I do not understand your questions.

Okay, I have a problem that after an unknown period of time the servos stop being set to values, I disabled the `#define DISABLE_PAUSE_RESUME disabled`, but the problem remains. The only time they start working is when I reset or send command `resumeWithInterrupts()`, can I send this command constantly, or can this cause problems with the library?

Leeeeerich, 2025-06-04:
Why closed? I have a problem, I see a problem, I fix it, but it is crunch.

ArminJo, 2025-06-04:
Who sends commands? 
And provide a **minimal** example, which shows the error.
Closed, since you do not prove, that the bug is in the library. I assume it is in your program!

Leeeeerich, 2025-06-04:
Well, if the reliability of the library is not the main thing, then there are no questions. Thank you for the wonderful library and documentation for it, I have been understanding it for a couple of days, and it is a crazy amount of work.

P.S. How can I prove to you about the problem, if the problem is repeated, but not from the moment of launch, but with different time intervals, there is a code example, in fact it is 4 lines of code from the library, that's all.

By the command I meant calling a method from the library, I already wrote from which object I take it. It's just that I call it remotely.

ArminJo, 2025-06-04:
Please specify FQBN (Fully Qualified Board Name).
        It is the first line in the compiler output in the Arduino IDE 2.x.
        The FQBN starts with the core name, which is essential bor bugfixing.
        E.g. for the bluepill board there are at least 2 prominent cores available, STMicroelectronics and stm32duino.
        **Without the FQBN** we are most likely not able to fix the the issue.**

Leeeeerich, 2025-06-11:
Processing nanoatmega328new (platform: ststm32; board: bluepill_f103c8; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 (19.2.0) > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)

> Please specify FQBN (Fully Qualified Board Name). It is the first line in the compiler output in the Arduino IDE 2.x. The FQBN starts with the core name, which is essential bor bugfixing. E.g. for [...]