ArminJo/ServoEasing #102: [Bug]: PCA9685 No I2C Adress 0x40 on Arduino Nano

nighteagle1974, 2025-09-05:
### Please specify "FQBN" (Fully Qualified Board Name). It is the first line in the compiler output in the Arduino IDE 2.x. E.g. "esp32:esp32:featheresp32:FlashFreq=80". The FQBN starts with the core name, which is essential for bugfixing. For example, at least two prominent cores are available for the Bluepill board, namely STMicroelectronics and STM32duino. Without the FQBN we are most likely not able to fix the the issue.

Arduino Nano 328P

### What IDE are you using?

Arduino 2.x IDE

### Example(s) you have checked while hunting the bug.

- [ ] ConsecutiveEasingsWithCallback
- [ ] ContinuousRotatingServo
- [x] OneServo
- [x] PCA9685_Expander
- [ ] Simple
- [ ] SimpleCallback
- [ ] TwoServos
- [ ] ThreeServos
- [ ] UnitTest
- [ ] Other - please specify below
- [x] I checked, whether at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] ConsecutiveEasingsWithCallback
- [ ] ContinuousRotatingServo
- [x] OneServo
- [x] PCA9685_Expander
- [ ] Simple
- [ ] SimpleCallback
- [ ] TwoServos
- [ ] ThreeServos
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [x] I use the [latest Arduino library version](../releases) and verified this!
- [x] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. Have installed the libs and add also the PinDefinitionsAndMore.h
2. Connect an Arduino Nano with the PCA9685 Standard Adress 0x40 all Jumper open
3. Check another PCA9685 sketch
`#include <Wire.h>
#include <Adafruit_PWMServoDriver.h>

Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver();

int currentPos[16];   // current positions for 16 servos numbered 0 - 15

void setup() {
  Serial.begin(9600);

  pwm.begin();
  pwm.setPWMFreq(60);

}

void loop() {

    slowMove0(7, 2, 800);
    slowMove0(6, 2, 500);
    slowMove0(5, 2, 500);
    delay(3000);
    slowMove0(7, 2, 100);
    slowMove0(6, 2, 100);
    slowMove0(5, 2, 100);
    delay(3000);
  

}`
All works fine.
5. Use Example PCA9685_ExpanderAndServo from ServoEasing
6. Uncoment in ServoEasing.h >> #define USE_PCA9685_SERVO_EXPANDER and also #define USE_SERVO_LIB
7.  Compile and upload all fine
8. Serial Monitor shows: Try to communicate with I2C device at address=0x40Error code=4. Communication with I2C was successful, but found no I2C device attached at address: 0x40
PCA9685 expander not connected

So the other sketch works without change anything on wiring the PCA to the Arduino Nano. It looks like something with the Pin-Mapping of the Nano in PinDefinitionsAndMore.h ? I2C on Nano is on A4 and A5.

### What happens?

No Connection to the PCA9685 Adress 0x40 with an Arduino Nano with Standard-Wiring i2c on Pin A4&A5

### The output which indicates the error happened.

```shell
Try to communicate with I2C device at address=0x40Error code=4. Communication with I2C was successful, but found no I2C device attached at address: 0x40
PCA9685 expander not connected
```

### What were you expecting to happen?

Found PCA9685

### Final checklist for the bug report.

- [x] I have **read** the README.md file thoroughly
- [x] I have browsed the examples for one that matches my use case.
- [x] The title of the issue is helpful and relevant.

nighteagle1974, 2025-09-06:
Hello,
 the PullUps are on the Adafruit PCA9865 Board. It is logical - because another lib works fine with the PCA9865. So there is no hardware or wiring issues cause the not detection of the Board.

ArminJo, 2025-09-06:
Did you check the power supply.
My problem is that I could reproduce the error once but after enabling the CPU pull-ups I cannot reproduce it any more, even after disabling the CPU pull-ups again.

nighteagle1974, 2025-09-06:
Hmmm.... curious...With the mormal Lib no problem.I have only one Nano Servo on the Board for testing and 5V 2A Supply the nano need 100mA.Von meinem/meiner Galaxy gesendet

nighteagle1974, 2025-09-06:
An idea... what about the speed of i2c with your lib instead of adafruit lib?Von meinem/meiner Galaxy gesendet

nighteagle1974, 2025-09-06:
Have fresh install on laptop with arduino 2.3.6Install the libs and use the example.Compule Error says there is no member in class namend initializedAndCheckI2CConnection...Did you mean InitializedAndCheck....Upper Letter "I" but in the ServoEasing both members... bool type, one with I and one with iHave Scrennshot but the github mail denied access. Von meinem/meiner Galaxy gesendet

nighteagle1974, 2025-09-06:
It looks like you have changed something from yesterday to today. In the new ServoEading.hpp is only one InitializeAndCheck with upper case. In the old file two of them with upper and lower case.Von meinem/meiner Galaxy gesendet

nighteagle1974, 2025-09-06:
Have use Arduino I2C Scanner and it found also the PCA9675 Board. Something is bugy wirh your lib on detecting routineVon meinem/meiner Galaxy gesendet

nighteagle1974, 2025-09-06:
Have figured out the problem. It is the speed of the i2c bus you set.I set it to 100kHz and now all works.Have set it quick dirty in line number 247setClock(I2C_ Clock_Frequency)TosetClock(100000)Von meinem/meiner Galaxy gesendet

nighteagle1974, 2025-09-07:
> Thanks for reporting, I modified the error message to cover this problem.

I think an extra define with the speed in the ino is fine. Standard 400kHz and if someone want can test up to 1MHz.So nobody struggle on parasitaer breadbord design for the first tests.. Von meinem/meiner Galaxy gesendet

nighteagle1974, 2025-09-08:
Hello, again, so i have looked deeper in the lib and it is the best i ever seen . it works brillant! So many thanks for this fantastic work!

So some question agsain  - is there a function to deactivate the servo - so it means no power to the servo? So if i detach(); it then it has power and hold its position but if i unconnect and reconnect to the board after detach(); then it is not hold any position means it have power loss. Don't know if this is an servo-specific thing or the PCA change something if connection to servo lost?

ArminJo, 2025-09-09:
detach() works for my SG90 servos! The output is constant low in this case.
And I fixed a serious reconnect() bug, so use the new version from this repo.